FreeRTOS/FreeRTOS-Kernel #2: Function to check if is inside a critical section

0x3333, 2019-10-14:
I'm porting FreeRTOS to ChibiOS HAL OSAL interface, and one of my needs is a function to tell if I'm inside a critical section or not.

Currently, I'm adding a function to portmacro.h in the architectures I'm working on, which is ARM Cortex Mx. But I need to maintain a fork just because of that function.

Could possibly this functionality be added to FreeRTOS? I can provide a `GCC/ARM_CMx` Pull Request.

Currently implementation for `GCC/ARM_CM3`:
```c
portFORCE_INLINE static BaseType_t xPortIsCriticalSection( void )
{
uint32_t ulReturn;

	__asm volatile
	(
		"	mrs %0, basepri	"
		:"=r" ( ulReturn )
	);

	return (ulReturn > 0) ? pdTRUE : pdFALSE;
}
```
Basically, I'm reading the `basepri` register, in the `CM0` case, I read the `PRIMASK` register.

Another option would be expose the `uxCriticalNesting` variable, as it is incremented every call to `vPortEnterCritical` and decremented on `vPortExitCritical`.

RichardBarry, 2019-10-14:
I will add this in to the SVN repo, under the existing xPortIsInsideInterrupt() utility function.

closed 2019-10-14 by RichardBarry

RichardBarry, 2019-10-14:
The code you post returns whether or not interrupts are masked or not, rather than whether you are inside a critical section or not - is that what you want/need?  There are other ways interrupts could get masked other then entering a critical section.

0x3333, 2019-10-14:
Technically speaking you are right. I need to know if I'm inside a critical section or not, more specifically if `vPortEnterCritical` was called or not. I believe the best way is to use the `uxCriticalNesting`, checking if it is `0` or not.

In my case, I'm checking if a function has been called inside a critical section, not if interrupts are masked or not.

In ChibiOS there are some functions(ending in `I`) that must be called from a critical section, and when assertions are enabled I need to check. That is why I need to know.

RichardBarry, 2019-10-14:
So would it be better to add something into the port layer along the lines of:

UBaseType_t uxGetCriticalNestingDepth( void );

that returns the critical nesting count depth - zero meaning you are not in a critical section?

0x3333, 2019-10-14:
Exactly. But the problem is that it is port dependent.

0x3333, 2019-10-15:
A function like `UBaseType_t xIsInsideCriticalSection( void );` would be more generic, usable in some ports that don't you nesting.

0x3333, 2019-10-15:
Like:

```
portFORCE_INLINE UBaseType_t xIsInsideCriticalSection( void )
{
     return ( uxCriticalNesting > 0 ) ? pdTRUE : pdFALSE;
}
```

RichardBarry, 2019-10-15:
...except it can't be an inline function in a header as uxCriticalNesting is not accessible from there.  I was thinking more as per the following as all ports support critical section nesting:

UBaseType_t uxGetCriticalNestingDepth( void )
{
    return uxCriticalNesting;
}

This would have to be added into each port layer.  The variable that counts the nesting depth is not always called the same thing.  Which specific port are you using (including the compiler) as I can start with that one.

0x3333, 2019-10-15:
You are right. I'm using GCC ARM Cortex M0, M3 and M4. IIRC they use `uxCriticalNesting`.

0x3333, 2020-03-17:
Hi! Do you believe is possible to provide the `uxGetCriticalNestingDepth` function?

RichardBarry, 2020-03-17:
It is not that straight forward because this needs to be in the port layer, and there are a LOT of different port layers that would then need slightly different implementations and testing.  I would like to propose a slightly different idea.  tasks.c provides a mechanism that enables enables users to add their own functions in to the file without needing to change the source file - you do that by setting a [#define in the FreeRTOSConfig.h file then providing a header file called freertos_task_c_additions.h](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/tasks.c#L5303) - we could do something similar in the port.c files - that way the same code would be added to the bottom of each port.c file and it would allow you to add your own access function (or any other port specific function) into a header file and have it included in the file.

Thoughts?

0x3333, 2020-03-17:
I understand, sure it is a massive work. Your suggestion would be killer, much more flexible. I can implement only the ports I need.

josesimoes, 2020-05-04:
Hello!

I'm after this xPortIsCriticalSection in GCC ARM Cortex M0.
Any update on when this (or the workaround above) is available?

Thanks!