FreeRTOS/FreeRTOS-Kernel #6: How to know if an ISR is executing ?

edbek, 2020-01-13:
Is it possible to determine whether a method in FreeRTOS is being invoked from the context of an ISR (interrupt service request) or a task at runtime ? If this is not, will there be such a possibility in FreeRTOS in the future ? 
Thanks.

RichardBarry, 2020-01-13:
The registers you need to read to determine if you are in an interrupt or not depends on the chip - some ports provide a macro to do this and you can do it yourself in cases where the macro doesn't exist.  Which port (chip and compiler) are you using?

edbek, 2020-01-19:
I use chip i.MX RT1064 and design environment MCUXpresso IDE.
Thank you.

edbek, 2020-01-19:
Is it possible to make a list of macros to determine the location inside interrupts for all popular processors?

RichardBarry, 2020-01-19:
> I use chip i.MX RT1064 and design environment MCUXpresso IDE.
> Thank you.

In which case you can use [xPortIsInsideInterrupt()](https://sourceforge.net/p/freertos/code/HEAD/tree/tags/V10.2.1/FreeRTOS/Source/portable/GCC/ARM_CM4F/portmacro.h#l169).

RichardBarry, 2020-01-20:
In future please post support requests to https://forums.freertos.org - thanks.

closed 2020-01-20 by RichardBarry