FreeRTOS/FreeRTOS-Kernel #39: [Inquiry] Would lower-power ports benefit from portTICK_TYPE_ATOMIC

Vbrzeski, 2020-03-27:
Looking at the latest kernel I noticed these macros:
portTICK_TYPE_ENTER_CRITICAL()
portTICK_TYPE_EXIT_CRITICAL()
portTICK_TYPE_SET_INTERRUPT_MASK_FROM_ISR()
portTICK_TYPE_CLEAR_INTERRUPT_MASK_FROM_ISR(x)

After some more review it would save ~10-13 instructions with no optimization, and ~4-5 on full optimization. The benefit from this enhancement of course depends on how often this function is called, but it might be desirable for low-power targets. My target is not that stringent, but an MSP430 would see some benefit here for example.

In generalAPI design adding flexibility comes with added complexity, and the question is if this would be worthwhile. My target is fine, but this may be different for an MSP430.

There is also atomic.h that could be consolidated with a similar feature. My only concern is offering the port this much flexibility may run into determinism issues if it is poorly implemented on the port.

RichardBarry, 2020-03-30:
Sorry, I'm not sure I understand what you are asking.

Vbrzeski, 2020-03-30:
Apologies for the confusion,

For my particular processor I could implement portTICK_TYPE_ENTER_CRITICAL with an ATOMIC instruction which disables interrupts in the pipeline for the next few instructions.

Now I originally posted this as a feature request and changed my mind after some review. The instruction I am referencing isn't present on any low-power processors, and my architecture is not commonly used with FreeRTOS. The savings are also quite small (~4-5 instructions per call).

I will close the inquiry since it is confusing and I think the feature wouldn't be utilized in most use-cases.

closed 2020-03-30 by Vbrzeski