FreeRTOS/FreeRTOS-Kernel #42: [BUG]Port for CCS/MSP430X - possible race condition in macro portRESTORE_CONTEXT

rphh, 2020-03-31:
**Describe the bug**
There's possible race condition in the macro `portRESTORE_CONTEXT`.
This may destroy the stack of task, and later cause a jump to invalid memory address.

The analysis of the bug was very difficult, so the description will be very difficult, but I'll do my best.
If something is unclear, let me know.

**Target**
- Development board: MSP430FR5969 (http://www.ti.com/tool/MSP-EXP430FR5969)
- Instruction Set Architecture: MSP430X (the bug may occur when small or large memory model is used)
- IDE and version: Code Composer Studio Version: 9.2.0.00013
- Toolchain and version: cl430 MSP430 C/C++ Compiler v18.12.3.LTS, msp430-elf-gcc (Mitto Systems Limited - msp430-gcc 8.3.0.16) 8.3.0

**Host**
- Host OS: Debian
- Version: 9

**To Reproduce**
- Use project with uses `vPortYield` frequently

**Expected behavior**
- Race condition in context switching code, jump to invalid address

**Screenshots**
Debugging session, gdb:
[image]

**Additional context**
I'm trying to create FreeRTOS port for MSP430 and GCC with support for large memory model.

I try to reuse code from two ports:
* https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/portable/CCS/MSP430X
* https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/master/portable/GCC/MSP430F449

The CCS port supports small memory model and large memory model.
The GCC port supports only small memory model.

The CSS port uses different approach in context switching code.

Generally: the context switching code is a critical section, the interrupts should be disabled till final return.

In the CSS port two instructions are used: `pop sr`, which enables interrupts, and `reta` which pops the return address from the stack and jumps to it. The last instruction `reta` **IS NOT** in critical section.
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/portable/CCS/MSP430X/portext.asm#L65

In the GCC port one instruction is used: `reti`, which pops four bytes from stack and restores values of Status Register and Program Counter. The instruction `reti` is in critical section.
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/portable/GCC/MSP430F449/port.c#L126

Additional info: MSP430FR58xx, MSP430FR59xx, and MSP430FR6xx Family User's Guide

If the Timer A ISR fire up after instruction `pop sr` and before instruction `reta`, the instruction `reta` will never be executed, and the stack of the task will be broken.

Scenario:
1. One of the tasks calls `vPortYield` (see screenshot)
2. context of the task is stored using macro `portSAVE_CONTEXT`
3. another task is selected
4. macro `portRESTORE_CONTEXT` is called (see screenshot)
  1. Machine instructions are executed, till `pop sr` (see screenshot)
  2. Instruction `pop sr` enables interrupts.
  3. There's still return address on the stack, which should be consumed by instruction `reta` (see screenshot)
  4. Interrupts are enabled, so the CPU jumps to timer A ISR, which performs context switching again.
5. The task of unlucky task is broken.

**Possible fix**
This is the hard part. I try to use `reti` instruction in the `portRESTORE_CONTEXT` macro, it works fine in small memory model, but in large memory model it needs additional hacking, but it works too.

RichardBarry, 2020-04-01:
Really appreciate you taking the time to provide this detail.  It may take a little time to digest and fix.  Would also be great if, once everything is running smoothly, you could also open a pull request to upstream your enhanced port.

rphh, 2020-04-01:
I was wrong.

The race condition is possible and if it happen the stack won't be broken.
The

RichardBarry, 2020-04-01:
Lets say Task A is executing the SR/RETI sequence when a context switch occurs to Task B between the SR and RETI instructions.  When the context switches back to Task A, would it not return to the RETI instruction allowing Task A to continue from there?  Is that what you mean by "I was wrong"?

rphh, 2020-04-02:
If the race condition occur, between `sr` and `reta` instructions, then the stack won't be broken. The instruction reta will be executed next time.

This is no longer a bug, so I'm closing it.

closed 2020-04-02 by rphh