FreeRTOS/FreeRTOS-Kernel #53: [BUG] ListItem_t and MiniListItem_t violate strict aliasing rules

Hawk777, 2020-04-16:
**Describe the bug**
ISO C’s “strict aliasing” rule says (roughly speaking) that the same memory must only be accessed either as the type it really is, or via a pointer to `char`. For example, an `int` can be accessed as an `int` or via a `char*`, but not via a `short*`. FreeRTOS breaks this rule: an object of type `List_t` contains an object of type `MiniListItem_t` (specifically `xListEnd`), but sometimes it is accessed via pointers of type `ListItem_t *` (not `MiniListItem_t *`). An example of this is in `uxListRemove` where, if removing the first item in the list, `pxItemToRemove->pxPrevious` is a `ListItem_t *` which points at `xListEnd`, and that pointer is written through in order to set `pxItemToRemove->pxPrevious->pxNext`, in violation of the strict aliasing rule.

Using aggressive optimizations in GCC 9, this generates bad code in `xTaskResumeAll` where `pxTCB` is not reloaded on each iteration of the first loop iterating over `xPendingReadyList`, resulting in the same TCB being added to the ready list more than once.

This could easily be fixed. Simply delete the `listFIRST_ITEM_INTEGRITY_CHECK_VALUE`, `xItemValue`, `pxNext`, and `pxPrevious` members from `ListItem_t` and replace them with an instance of `MiniListItem_t` (which contains exactly those members, and therefore does not change the memory layout of the structure at all), and fix up the resulting compile errors (which involves changing some `ListItem_t *` pointers to `MiniListItem_t *`). This should be totally standards-compliant. Now all iteration happens using the `MiniListItem_t` type, so the strict aliasing rule is obeyed. As long as a pointer is definitely pointing at a real item (not `xListEnd`), casting from `MiniListItem_t *` to `ListItem_t *` is also legal, because that is a cast between a pointer to a structure and a pointer to its first member, which is legal in both directions.

I would be happy to submit a PR if you agree that this should be changed.

Technically speaking I believe the `Static` types also constitute violations of the strict aliasing rule (for example, FreeRTOS accesses an object of type `struct xSTATIC_TCB` through a pointer of type `struct tskTaskControlBlock *`), but so far I have not seen this to cause actual problems, and it seems less likely to do so since the object is never actually accessed via its real type.

**Target**
- Instruction Set Architecture: ARMv7-M
- Toolchain and version: GCC 9.3.0

**Host**
- Host OS: Linux
- Version: Kernel 5.4.28

RichardBarry, 2020-04-16:
I've not been able to replicate an issue, but I did try making the change you suggested and confirm all the tests are still passing.  It does however break the TCP/IP code as that is [incorrectly] accessing list members directly - so there will be TCP stack version Vs kernel version compatibility issues we will also have to manage.

Grateful if you can create the PR as you suggest so you can also check it fixes the issue you found.  Please also let us know the GCC command line you are using so we can add it to our tests too.

Hawk777, 2020-04-17:
With extraneous flags removed:

```
armv7m-none-eabihf-gcc-9.3.0 -O3 -ggdb -g3 -fno-common -fno-math-errno -fstrict-aliasing -std=c99 -nostdlib -mabi=aapcs -fno-exceptions -ffreestanding -mslow-flash-data -flto=jobserver -mfloat-abi=hard -mcpu=cortex-m7 -mfpu=fpv5-sp-d16 -mthumb
```

Similar flags (including LTO, of course) are also used at link time.

Hawk777, 2020-04-30:
I asked this in the PR, but as you closed it perhaps you don’t see those messages.

Is that coordination something that you would expect to happen at some point? What kind of timeframe would it involve?

I admit I never tried any such things (at such high optimization levels I’ve never had very good results with them anyway; required symbols tend to be optimized out of existence). However, OpenOCD, the only one I’m familiar with, appears to hardcode the memory layout of a list element, so since this patch doesn’t change the layout, it should be unaffected?

belchercw, 2021-03-01:
In case review comments are not being seen, I will repost my comment here as well:

I just spent an entire week trying to track down a crash in my project which turns out to be this bug. This is a very insidious issue that creates intermittent instability and is very difficult (read expensive) to track down. As the correct fix has been rejected, what is the proposed resolution, and how do we avoid others continuing to encounter the problem?

RichardBarry, 2021-03-01:
I don't think the proposed fix has been rejected - but first we need to update the +TCP code to use access functions/macros rather than accessing the list structure members directly.  

There are many kernel aware debug tools for FreeRTOS and changing structures or variable names often breaks them.  You can see that legacy in the code comments.  From your post it sounds like OpenOCD is using these structures directly so we need to check we are not breaking that or other debuggers too.  It might be an issue where the debugger is reading the symbol table directly rather than assuming a memory layout.

Hawk777, 2021-03-01:
OpenOCD assumes a memory layout, so it would *not* break (since my proposed fix does not change the memory layout). OpenOCD only reads the ELF symbol tables, not the DWARF data, so it knows about global variables but not struct members; the offset of the “next” pointer in the list structure is hardcoded.

belchercw, 2021-03-01:
My quick fix is to add `-fno-strict-aliasing`, which appears to solve the problem, though this has potential performance implications.  I see that gcc enables strict-aliasing optimizations at O2 and above, so I would expect this to be encountered frequently.   Not sure if this is isolated to gcc 9.x, but I am running 9.2.1, which came as part of the MCUXpresso distribution.

t-8ch, 2021-03-29:
FYI after running into issues with the hardcoded structure offsets in OpenOCD I sent a request to be able to use dynamic structure offsets
(via GDB): https://sourceware.org/pipermail/gdb/2021-March/049329.html

Viatorus, 2021-10-06:
Any updates on this fix?

paulbartell, 2021-12-14:
I spoke with @RichardBarry and we came to the conclusion that the simplest way to quiet this warning would be to add a configuration option to:

1. Exclude the existing definition for MiniListItem_t and
2. typedef ListItem_t and MiniListItem_t to the same type.

i will open a PR doing so shortly.