FreeRTOS/FreeRTOS-Kernel #66: Parenthesis around if-statement in macros

the78mole, 2020-05-29:
When macro is used within an if-else statement, it can lead to ambiguities, when not sourrounded by curly braces.

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/4b353bfd7aa009d553f31b4e47dfb49a9f3c75b1/portable/GCC/ARM_CM4F/portmacro.h#L93

It is a good practice and makes it more developer convenient if the following piece of code would be used:
```
#define portEND_SWITCHING_ISR( xSwitchRequired ) \
    do \
        { if( xSwitchRequired != pdFALSE ) portYIELD(); \
    } while(0)
```

Reference Issue: https://github.com/STMicroelectronics/STM32CubeF4/issues/20

alfred2g, 2020-08-26:
Hi @the78mole,

Did this macro caused you any issues ?
Also, could you please open a discussion thread for it in the FreeRTOS [support forums](https://forums.freertos.org/) to get a better insight ?

Alfred

the78mole, 2020-08-26:
Hi Alfred, 

indeed I got a compiler warning, when I used it last time. But since I'm in vacations and sticked to my smartphone, it's not as easy to raise a discussion in support forums.

Regards,
Daniel

cobusve, 2020-09-28:
Fixed in PR #138

closed 2020-09-28 by cobusve