FreeRTOS/FreeRTOS-Kernel #68: [BUG]: MPU not enabled for CM7 core.

rahmanih, 2020-06-01:
Hi,

**Describe the bug**

the "ARM_CM4F_MPU" port is usable by either the CM4 based CMUs or CM7 ones (except the r0p1 version).

in this port the _*portEXPECTED_MPU_TYPE_VALUE*_  is forced to 8 regions, while according to the [ARM CM7 documentation](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0489f/BIHBEJIC.html), the MPU region can be 16 regions depending on the implmentation.

For some targets like the "STM32H7",FreeRTOS will *silently fail* in [this check](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/4b353bfd7aa009d553f31b4e47dfb49a9f3c75b1/portable/IAR/ARM_CM4F_MPU/port.c#L517)  and the MPU won't be correctly enabled as the regions in this MCU are 16 not 8.

**Target**
- STM32H743 eval board.
- Instruction Set Architecture: [e.g. ARMv7]
- IDE and version: [EWARM(but valid of other IDEs)]
- Toolchain and version: [8.30.1]

**Host**
- Host OS: [Windows]
- Version: [10]

**To Reproduce**
- Run a FreeRTOS/MPU based application.
- verify that the MPU is not correctly configured.

regards
Haithem.

yuhui-zheng, 2020-06-09:
Thank you for raising this issue. 

For our side:
I think it's a good practice to always have an else branch. And in this case, adding a #warning for ```portMPU_TYPE_REG != portEXPECTED_MPU_TYPE_VALUE ``` would be something quick dirty we can do to at least ease the pain.

For your side:
I'm not sure how far you went with your project. But in case you may need to jump start on 16-entry MPU...
- [```prvSetupMPU()```](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/4b353bfd7aa009d553f31b4e47dfb49a9f3c75b1/portable/IAR/ARM_CM4F_MPU/port.c#L507) This function as you are already aware of, is for setting up MPU entries for none task specific regions. If your linker script has the same [symbols](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/4b353bfd7aa009d553f31b4e47dfb49a9f3c75b1/portable/IAR/ARM_CM4F_MPU/port.c#L509-L514) and section size/alignment, you may likely be able to just use this part with minimum modification. 
- And since now we have more entries, these can be used (partially or however suits your application scenario) for either task specifics or global settings. If global, could append ```prvSetupMPU()```. If task specific, need to save settings in task context -- related functions, namely [```vPortRestoreContextOfFirstTask()```](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/4b353bfd7aa009d553f31b4e47dfb49a9f3c75b1/portable/IAR/ARM_CM4F_MPU/portasm.s), and [```vPortStoreTaskMPUSettings()```](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/4b353bfd7aa009d553f31b4e47dfb49a9f3c75b1/portable/IAR/ARM_CM4F_MPU/port.c#L594).

rahmanih, 2020-06-09:
Hi @yuhui-zheng,

thanks for the answer, but I didn't really catch your comment above, could you please explain further?
Do you mean that the [prvSetupMPU()](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/4b353bfd7aa009d553f31b4e47dfb49a9f3c75b1/portable/IAR/ARM_CM4F_MPU/port.c#L507) doesn't need to be updated to support the 16 regions MPU?

sorry for the disturbance.

regards
Haithem.

yuhui-zheng, 2020-06-09:
Before more words to confuse you... Supporting 16-entry MPU out-of-box is of our interest, though I don't have a concrete timeline to share when typing this line. The bullet points are mostly for "in case you need it yesterday" and you were planning to customize existing MPU port. 

Expanding comments on ```prvSetupMPU()``` a little bit. (In your first post you linked IAR/CORTEX_M4F_MPU port, so that below is specific to this port.)

The other part to MPU initialization is how memory is partitioned in your application (with linker script). As an example, [```CORTEX_MPU_STM32L4_Discovery_GCC_IAR_Keil/Projects/IAR```](https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS/Demo/CORTEX_MPU_STM32L4_Discovery_GCC_IAR_Keil/Projects/IAR) this project uses this [linker script](https://github.com/FreeRTOS/FreeRTOS/blob/master/FreeRTOS/Demo/CORTEX_MPU_STM32L4_Discovery_GCC_IAR_Keil/Projects/IAR/stm32l475xx_flash.icf) along with IAR/CORTEX_M4F_MPU port. ```__privileged_functions_start__``` is set to address 0x08000000, and ```__privileged_functions_end__```  is set to address 0x08007FFF. So that [these lines](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/4b353bfd7aa009d553f31b4e47dfb49a9f3c75b1/portable/IAR/ARM_CM4F_MPU/port.c#L529-L538) set up an MPU entry to grant privilege mode read-only access to functions in memory address range of [```__privileged_functions_start__```, ```__privileged_functions_end__```]. (Also executable, see XN bit in MPU_RASR register.)

So if to setup a new project with STM32H743 evaluation board, may need to get a reasonable memory partition as well. Hope this clarifies a little bit. 

Regards,
Yuhui

rahmanih, 2020-06-16:
Thanks I'm closing this issue now.

regards
Haithem.

closed 2020-06-16 by rahmanih