FreeRTOS/FreeRTOS-Kernel #90: [BUG] Uncrustify breaks inline asm code for Renesas CC-RX compiler

NoMaY-jp, 2020-07-17:
Hello,

I notice the following two problems caused by Uncrustify.

(1) There are some mistakes of indent.
(2) Single line comment starting with ';' is moved to the next line. This causes compile error.

For example, following changes had been made.

portable/Renesas/RX600v2/port.c

Before:

```
#pragma inline_asm prvYieldHandler
static void prvYieldHandler( void )
{
...omit...

    /* Save the FPSW and accumulators. */
    MVFC    FPSW, R15
    PUSH.L  R15    <---- HERE and BELOW.
    MVFACGU #0, A1, R15
    PUSH.L  R15
    MVFACHI #0, A1, R15
    PUSH.L  R15
    MVFACLO #0, A1, R15 ; Low order word.    <---- HERE
    PUSH.L  R15
    MVFACGU #0, A0, R15
    PUSH.L  R15
    MVFACHI #0, A0, R15
    PUSH.L  R15
    MVFACLO #0, A0, R15 ; Low order word.    <---- HERE
    PUSH.L  R15

...omit...
}
```

After:

```
#pragma inline_asm prvYieldHandler
static void prvYieldHandler( void )
{
...omit...

    /* Save the FPSW and accumulators. */
    MVFC FPSW, R15
        PUSH.L R15     <---- mistakes of indent. HERE and BELOW.
        MVFACGU # 0, A1, R15
        PUSH.L R15
        MVFACHI # 0, A1, R15
        PUSH.L R15
        MVFACLO # 0, A1, R15;
    Low order word.    <---- This should be comment.
       PUSH.L R15
        MVFACGU # 0, A0, R15
        PUSH.L R15
        MVFACHI # 0, A0, R15
        PUSH.L R15
        MVFACLO # 0, A0, R15;
    Low order word.    <---- This should be comment.
       PUSH.L R15

...omit...
}
```

Best regards,
NoMaY

NoMaY-jp, 2020-07-17:
Hello,

I think that one possible solution for type (1) is as following. (But I can't guess what causes type (2).)

One possible solution for type (1):

(Step 1) Revert changes made by Uncrustify in case of source files for CC-RX compiler.
(Step 2) Replace single line comment starting with ';' by "/*" ~ "*/".
(Step 3) Execute Uncrustify again.

Best regards,
NoMaY

lundinc2, 2020-07-23:
Thank you for bringing this to our attention @NoMaY-jpm, I will begin investigating this issue.

NoMaY-jp, 2020-07-23:
Thank you for your attention @lundinc2. I noticed more inline asm codes which cause compile error. The below 1st and 2nd instructions are used in the Renesas RX600v2 port layer etc. The 3rd ~ 6th instructions will be used new Renesas RX700v3 DPFPU port layer which is being prepared by me for new pull request. Unfortunately white spaces around hyphen causes compiler error when Renesas CC-RX compiler is used.

Before Style:Uncrustify - Compiler error is not caused.
```
    POPM R1-R15
    PUSHM R1-R14

    DPOPM.L DPSW-DECNT
    DPOPM.D DR0-DR15
    DPUSHM.D DR0-DR15
    DPUSHM.L DPSW-DECNT
```
After Style:Uncrustify - Compiler error is caused.
```
    POPM R1 - R15
    PUSHM R1 - R14

    DPOPM.L DPSW - DECNT
    DPOPM.D DR0 - DR15
    DPUSHM.D DR0 - DR15
    DPUSHM.L DPSW - DECNT
```

alfred2g, 2020-08-19:
Reverting files are merged.
I will close this issue, @NoMaY-jp, please reopen if you still see any issue.

closed 2020-08-19 by alfred2g