FreeRTOS/FreeRTOS-Kernel #91: [BUG] - uncrustify breaks formatting for inline assembly in ATmega port

feilipu, 2020-07-17:
**Describe the bug**
Uncrustify breaks formatting for inline assembly code in port.c and portmacro.h. This makes the code unreadable.

**Target**
ThirdParty/GCC/ATmega

**Expected behavior**
Human legible code.

** Current situation**
[image]
[image]

lundinc2, 2020-07-24:
Hello @feilipu,

Thank you for submitting this report. It seems uncrustify has caused several issues. We will investigate and report back with a fix.

alfred2g, 2020-08-24:
Code is now looking readable to some people at least :) 
closing the issue

closed 2020-08-24 by alfred2g