FreeRTOS/FreeRTOS-Kernel #93: [BUG] xEventGroupClearBitsFromISR execution delay

VladimirUmek, 2020-07-17:
Below is a simple example that causes event group clear operation (from ISR) to execute after event group set operation:
```
/* Interrupt handler (ISR context) */
void Peripheral_IRQHandler (void) {
  BaseType_t rval;

  rval = xEventGroupClearBitsFromISR (hEventGroup, (EventBits_t)0x00000001U);
}

/* Test task */
void Test (void *arg) {

  hEventGroup = xEventGroupCreate();

  bits_a = xEventGroupSetBits (hEventGroup, (EventBits_t)0x00000001U);
  // After execution: bits_a == 1

  // Peripheral_IRQHandler was triggered and executed (ISR context)

  bits_b = xEventGroupSetBits (hEventGroup, (EventBits_t)0x00000001U);
  // After execution: bits_b == 0
}
```
In a Test task, Event Group object is created and its LSB is set using xEventGroupSetBits .

Next, peripheral interrupt is fired and Peripheral_IRQHandler is executed, which calls xEventGroupClearBitsFromISR. This function registers vEventGroupClearBitsCallback for execution and moves timer service/daemon task to ready state.

After interrupt handler returns, xEventGroupSetBits is called again, setting LSB. This function calls vTaskSuspendAll, sets the LSB and before it returns, calls xTaskResumeAll - at this point, timer service/deamon task gets its chance to execute and calls vEventGroupClearBitsCallback which clears the Event Group LSB. Bits returned by pxEventBits->uxEventBits are therefore invalid.

This was verified on an Cortex-M target.

See
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/c273690296eb1767466c6276dd7d5a28ea20c37a/event_groups.c#L502
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/c273690296eb1767466c6276dd7d5a28ea20c37a/timers.c#L564
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/c273690296eb1767466c6276dd7d5a28ea20c37a/timers.c#L756
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/c273690296eb1767466c6276dd7d5a28ea20c37a/event_groups.c#L680
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/c273690296eb1767466c6276dd7d5a28ea20c37a/event_groups.c#L532
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/c273690296eb1767466c6276dd7d5a28ea20c37a/event_groups.c#L623

A simple workaround is to just use portYIELD_FROM_ISR:
```
/* Interrupt handler (ISR context) */
void Peripheral_IRQHandler (void) {
  BaseType_t rval;

  rval = xEventGroupClearBitsFromISR (hEventGroup, (EventBits_t)0x00000001U);

  if (rval == pdPASS) {
    portYIELD_FROM_ISR (pdTRUE);
  }
}
```

RichardBarry, 2020-08-06:
This looks like an inconsistency in the API as the normal "xHigherPriorityTaskWoken" parameter is not present from the "FromISR" version of the API.  It is difficult to change the API as it would break pre-existing applications that use it.  We could update the documentation to demonstrate calling the "yield from ISR", as per your code snippet, or introduce a new function with the omitted parameter.  A new API would be inconsistent in its naming, so I'm favoring documentation changes rather than code changes.

VladimirUmek, 2020-08-18:
Argument for not breaking the API is understandable and proposed solutions are also acceptable. The main part is that the problem must to be understood by developers so at least documentation update is necessary.

closed 2021-12-28 by n9wxu