FreeRTOS/FreeRTOS-Kernel #97: should tasks run in u mode for riscv?

gxflying, 2020-07-20:
hi sir, I am learning about the RTOS source code for riscv now, and I found that all the code runs in M-mode。
I think the normal task maybe run in U-mode is better for securety concerns， 
how do you think ?  hope your reply !
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/6199b72fbf57a7c5b3d7b195a3bd1446779314cd/portable/GCC/RISC-V/portASM.S#L421

yuhui-zheng, 2020-07-21:
Hi @gxflying,

You are correct that for now our RISC-V ports only run in M-mode, and we have an item in our todo list to support U-mode. If you are already experimenting with U-mode, happy to chat about the details.

gxflying, 2020-07-21:
it‘s great to hear about this !
and any progress from mine I will comment here , thanks~

yuhui-zheng, 2020-07-22:
@gxflying  You are welcome!

Kindly be aware that there may be relevant PRs on this topic recently. Feel free to pop in and comment as well. I'm closing this issue for now.

closed 2020-07-22 by yuhui-zheng