FreeRTOS/FreeRTOS-Kernel #100: get next instuction by adding 4 may result in wrong address when supporting "c" extension

gxflying, 2020-07-21:
Hi sir,  in the portASM.S for riscv, I found that when traped by a synchronous exception,  the next address to fetch instrustions
is calaulated by adding 4 ?!  this may not always true when the core supporting "c" extension and the current execting instruction 
is a compress instruction which is two bytes width,   then the next address should be  **pc + 2** !
it's better to get the current instruction value, and check weather it's a "c" instr or NOT , and then decide to add 4 or 2
how do you think about this ?
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/6199b72fbf57a7c5b3d7b195a3bd1446779314cd/portable/GCC/RISC-V/portASM.S#L229

RichardBarry, 2020-07-21:
Thanks for providing this information - currently we don't support the 'C' extension but if this is something of interest to you then we can add it to our backlog.  Alternatively if you wish to see support sooner then I would encourage you to submit a pull request with the necessary modifications.

closed 2020-07-21 by RichardBarry

gxflying, 2020-07-22:
Hi, @RichardBarry I may not agree with you.
see  risc-v demo below, all the gcc options are "-march=rv32imac",  so at least the demo developers did NOT known the information that  "currently we don't support the 'C' extension ".
https://github.com/FreeRTOS/FreeRTOS/blob/8e36bee30eef2107e128edb58e83ee46e8241a91/FreeRTOS/Demo/RISC-V_RV32_SiFive_HiFive1-RevB_FreedomStudio/.cproject#L88
I think there two ways for it:
1.  explicitly claim that “c” extension NOT support yet somewhere easy to be seen
2. I will try to PR as what I said in the first comment

bluewww, 2020-07-22:
Indeed the addition by the instruction size needs to be different for each trap handler. In this case the `addi a1, a1, 4` should go under `test_if_environment_call`. The instruction size of `ecall` is always four.