FreeRTOS/FreeRTOS-Kernel #98: Could provide a .uncrustify.cfg file in root directory

YuguoWH, 2020-07-21:
late commits of uncrustify has made some code unreable, and new code submit could also bring code style problems. Therefore I would suggest puting a .uncrustify.cfg in root directory. So pull requests could align to a specify code style before they are merged, and also presenting a chance to find bugs ahead.

lundinc2, 2020-07-23:
Hello @YuguoWH,

I am in the process of determining a proper locating for the uncrustify associated with the kernel.

Thanks

lundinc2, 2020-07-24:
Hello @YuguoWH,

I have updated the Readme in this repository to specify where the uncrustify config can be found. We decided to put it in FreeRTOS/FreeRTOS, under the tools directory, in to keep this repository small. Thank you for bringing this to our attention and please re-open this issue for further help.

closed 2020-07-24 by lundinc2