FreeRTOS/FreeRTOS-Kernel #119: [BUG] Case sensitivity issue with MSVC_MinGW portmacro.h

XaviDCR92, 2020-08-13:
**Describe the bug**
The `MSVC_MinGW` version of portmacro.h causes compilation errors when cross-compiling from GNU/Linux to Windows (both x86 and x86_64) since the header files it includes are written in sentence case e.g.: `Windows.h` whereas the `mingw-w64-x86-64-dev` package distributes them in lowercase e.g.: `windows.h`.

While this would not be an issue when compiling from Windows, it does affect those cross-compiling from any case-sensitive operating system such as GNU/Linux to Windows.

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/4bde12f223c5eb14e3dc75518b563d94d722fa03/portable/MSVC-MingW/portmacro.h#L31
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/4bde12f223c5eb14e3dc75518b563d94d722fa03/portable/MSVC-MingW/portmacro.h#L32

**Target**
- Development board: N/A
- Instruction Set Architecture: x86_64-w64-mingw32
- IDE and version: N/A
- Toolchain and version: x86_64-w64-mingw32-gcc (GCC) 7.3-win32 20180312

**Host**
- Host OS: x86_64 GNU/Linux Kubuntu
- Version: 20.04 LTS

**To Reproduce**
Cross-compile the Windows port of FreeRTOS from GNU/Linux or any other case-sensitive operating system that provides the `mingw-w64-x86-64-dev` package or similar.

**Expected behavior**
One or more compilation errors are triggered by `x86_64-w64-mingw32-gcc` each time `portmacro.h` is either directly or indirectly included by any other source file. The following description is provided by the compiler:

```
FreeRTOS-Kernel/portable/MSVC-MingW/portmacro.h:31:10: fatal error: Windows.h: No such file or directory
 #include <Windows.h>
          ^~~~~~~~~~~
```

**Screenshots**
N/A

**Additional context**
The solution is rather straightforward: modify the lines mentioned above so lowercase names are used instead. This would not affect those already compiling from Windows and would allow cross-compiling to Windows from case-sensitive platforms such as GNU/Linux.

mingyue86010, 2020-08-27:
Thank you for reporting this and explaining in detail. 

We've added the fix here: https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/144

Please let me know if you have any further concerns.

XaviDCR92, 2020-08-28:
Thank you very much for your feedback. I have approved the changes from the pull request.

closed 2020-08-28 by XaviDCR92