FreeRTOS/FreeRTOS-Kernel #139: [BUG] The newly added GCC/RX200 port is wrong.

NoMaY-jp, 2020-08-26:
**Describe the bug**

The port should be for RX200 group MCUs using RXv1 CPU core but without FPU. RXv1 CPU core has only one accumulator. (On the other hand, RXv2 and RXv3 CPU cores has two accumulators.) Nevertheless the port attempts to save/restore FPSW(FPU's PSW) and two accumulators(using RXv2 CPU core instructions)! (Please compare GCC/RX200/port.c and Renesas/RX200/port.c.)

Conversation with source code is here.

Update Renesas GCC compiler ports #135
[https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/135#issuecomment-680609307](https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/135#issuecomment-680609307)

Best regards,
NoMaY

NoMaY-jp, 2020-08-26:
Hello,

The following table describes which port layer in FreeRTOS v10.3.1 should be used and it shows the reason why this mistake is caused.
```
RX MCU Group        Port Layer
                    CC-RX               GNURX           ICCRX
RX111               Renesas/RX100       GCC/RX100       IAR/RX100
RX113               Renesas/RX100       GCC/RX100       IAR/RX100
RX210               Renesas/RX200       N/A             N/A
RX62N,RX621         Renesas/RX600       GCC/RX600       IAR/RX600
RX630               Renesas/RX600       GCC/RX600       IAR/RX600
RX63N,RX631         Renesas/RX600       GCC/RX600       IAR/RX600
RX64M               Renesas/RX600v2     GCC/RX600v2     IAR/RXv2
RX71M               Renesas/RX600v2     GCC/RX600v2     IAR/RXv2
...
RX13T               Renesas/RX600       GCC/RX600       IAR/RX600
RX231               Renesas/RX600v2     GCC/RX600v2     IAR/RXv2
RX23W               Renesas/RX600v2     GCC/RX600v2     IAR/RXv2
RX65N               Renesas/RX600v2     GCC/RX600v2     IAR/RXv2
```
Best regards,
NoMaY

NoMaY-jp, 2020-08-27:
Hello,

Probably the following table will add more complexity but port layer should be used as follows.
```
RX MCU  CPU     FPU         FPU         Port Layer
Group   Core    (Single     (Dobule     CC-RX                   GNURX               ICCRX            
                Precision)  Presision)                                                               
RX72N   RXv3    Yes         Yes         Renesas/RX700v3_DPFPU   GCC/RX700v3_DPFPU   IAR/RX700v3_DPFPU
RX72M   RXv3    Yes         Yes         Renesas/RX700v3_DPFPU   GCC/RX700v3_DPFPU   IAR/RX700v3_DPFPU
RX66N   RXv3    Yes         Yes         Renesas/RX700v3_DPFPU   GCC/RX700v3_DPFPU   IAR/RX700v3_DPFPU
RX72T   RXv3    Yes         No          Renesas/RX600v2         GCC/RX600v2         IAR/RXv2         
RX66T   RXv3    Yes         No          Renesas/RX600v2         GCC/RX600v2         IAR/RXv2         
```
Best regards,
NoMaY

ravibhagavandas, 2020-08-28:
@NoMaY-jp 
Looking at the demos we have only one demo for RX 200 V1 MCU (RX210) which is for Renesas compiler: [RX200_RX210-RSK_Renesas](https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS/Demo/RX200_RX210-RSK_Renesas).
We have demos for RX200 v2 MCU (RX 231) for both Renesas and GCC compilers: [RX200_RX231-RSK_Renesas_e2studio](https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS/Demo/RX200_RX231-RSK_Renesas_e2studio) and [RX200_RX231-RSK_GCC_e2studio_IAR](https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS/Demo/RX200_RX231-RSK_GCC_e2studio_IAR).
So, should the port be actually supporting RX200 v2 and be renamed as GCC/RX200v2 ? Also I see that demo project for RX231 in e2studio/GCC uses RX600v2 port. I think this needs to be also changed to RX200v2 for GCC.

NoMaY-jp, 2020-08-28:
Hello @ravibhagavandas,

> So, should the port be actually supporting RX200 v2 and be renamed as GCC/RX200v2 ? Also I see that demo project for RX231 in e2studio/GCC uses RX600v2 port. I think this needs to be also changed to RX200v2 for GCC.

I think that removing the newly (mistakenly) added GCC/RX200 port is enough for this issue. I proposed removing this port to Renesas Electronics Corporation yesterday via Japanese user forum and today I received an answer that they agree with me. So, I'd like to post a pull request to remove this port.

On the other hand, renaming this port isn't enough for all complexity of Renesas RX ports because there are many other mismatch among MCU group names and CPU core versions and port names as I wrote in the tables.

I think that in the past the following port names were easy to understand that which port should be used.
```
RX100
RX200
RX600
RX600v2
```
But I think that as of today the following port names are better but I think that renaming all ports isn't worth to do so. (But... I'd like to rename IAR/RXv2 to IAR/RX600v2 because only this port name isn't consistent with other port names...)
```
RXv1_NOFPU_WITH_TICKLESS_MODE   <-- current RX100 port
RXv1_NOFPU                      <-- current RX200 port (for CC-RX only )
RXv1_FPU                        <-- current RX600 port
RXv2_FPU                        <-- current RX600v2 port (no MCU groups using RXv2 without FPU as of today)
RXv3_NODPFPU                    <-- this port will be the same as RXv2_FPU port (no demos as of today)
RXv3_DPFPU                      <-- current RX700v3_DPFPU port
```
And there is a possibility like the following ports in the future.
```
RXv2_FPU_WITH_TICKLESS_MODE
RXv3_NODPFPU_WITH_TICKLESS_MODE
RXv3_DPFPU_WITH_TICKLESS_MODE
RXv1_FPU_WITH_MPU_MODE
RXv2_FPU_WITH_MPU_MODE
RXv3_NODPFPU_WITH_MPU_MODE
RXv3_DPFPU_WITH_MPU_MODE
and combination with tickless mode and MPU(Memory Protection Unit).
```
Of course, using #ifdef ~ #endif can reduce ports as follows.
```
RXv1
RXv2
RXv3
```
By the way, Renesas Electronics Corporation doesn't uses the following names in their documents.
```
RX200v2
RX600v2
RX700v3
```
They use names like the following words. (RXv3 CPU core has one more option but it is not a topic in this issue.)
```
RXv1 with FPU
RXv1 without FPU
RXv2 with FPU
RXv3 with DPFPU
RXv3 without DPFPU
```
Best regards,
NoMaY

NoMaY-jp, 2020-08-28:
Hello @ravibhagavandas,

I posted a pull request which removes GCC/RX200 port.

Best regards,
NoMaY

HirokiIshiguro, 2020-10-03:
Hi, NoMay-san, and AWS team. This is Ishiguro from Renesas software team. 
We should keep current port (skip #146)”. Renesas thinks very small impact without #146.
And #146 seems causing conflicting with master branch.
So we should close #146 in current situation.

closed 2020-10-04 by cobusve