FreeRTOS/FreeRTOS-Kernel #154: Unable to compile FreeRTOS-Kernel files with Keil Compiler Version 6

rxa1031, 2020-09-02:
Hi,

Using STM32CubeMX, I've generated a project for understanding how to create tasks. The (possibly) used FreeRTOS version is 10.2.1 with CMSIS-RTOS version 2.00

When using Keil ARM Compiler version 6 to compile the code, I am observing several concerns, possibly because the FreeRTOS-Kernel code is still compilable with help of Keil ARM Compiler Version 5

The files under question (port.c and portmacro.h) are available under folder:
https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/master/portable/RVDS/ARM_CM4F

I have made modifications to the files and brought them up to a level wherein the compiler is now not throwing errors. I however need your help with _how to correctly use the extern variables inside the ASM code_. I am also concerned about how to handle C variables in assembly.

I have used the **__CC_ARM macro** to separate out the assembly code which can be compiled with Keil Compiler Version 5 from the one that can be compiled with Keil Compiler Version 6.

I hereby request you to please review the files and help me with an updated file ASAP (asit is crucial to my project).

Please help.

Thanks,
Rajeev

rxa1031, 2020-09-02:
[port.zip](https://github.com/FreeRTOS/FreeRTOS-Kernel/files/5160483/port.zip)

rxa1031, 2020-09-02:
Hi,

In file portmacro.h function vPortClearBASEPRIFromISR() if I try to replace:
	_msr basepri, #0_

with:
	___set_BASEPRI(0);_

And include one of the two header files in which the function is declared, the below errors are observed:

```
../Middlewares/Third_Party/FreeRTOS/Source/portable/RVDS/ARM_CM4F\portmacro.h(35): warning: In file included from...
../Drivers/CMSIS/Include\cmsis_gcc.h(527): error: redefinition of '__get_FAULTMASK'
__STATIC_FORCEINLINE uint32_t __get_FAULTMASK(void)
                              ^
../Drivers/CMSIS/Include/cmsis_armclang.h(512): note: previous definition is here
__STATIC_FORCEINLINE uint32_t __get_FAULTMASK(void)
```

Please help with fixing the issue.

Regards,
Rajeev

rxa1031, 2020-09-16:
@n9wxu @jeremyerdmann @RichardBarry @feilipu @syntroniks Please help with above issue. I am unable to deliver project updates as I am unable to compile FreeRTOS with Keil Compiler version 6.

RichardBarry, 2020-09-16:
Grateful if you can please re-post in the support forum here:  https://forums.freertos.org, where we will be happy to assist, and everybody else will also see the responses in case they have the same issues in the future.

rxa1031, 2020-09-18:
@RichardBarry

Thanks for your response.

As per your suggestion, I've uploaded my concern to the Forum page as well. Below is the link.

https://forums.freertos.org/t/unable-to-compile-freertos-kernel-files-with-keil-compiler-version-6/10697

Regards,
@rxa1031

aggarg, 2020-09-18:
Here is a project that I generated using CubeMX and updated to use GCC port files which are needed for Keil Compiler Version 6:
[stm32h753.zip](https://github.com/FreeRTOS/FreeRTOS-Kernel/files/5246366/stm32h753.zip)

Thanks.

rxa1031, 2020-09-18:
Dear Gaurav,

How Did you instruct STM32CubeMX to use the GCC (folder) while generating code? Is there a way to tell STM32CubeMX to use GCC and not use RVDS? If I manually make a change to code, I am concerned that the next time I generate code with help of same IOC file, it may delete the GCC folder and instead add the RVDS folder.

I used the STM32CubeMX version 6.0.1 available on my laptop and created a project for STM32H753BIT6.

I compared the generated IOC file with the file you shared, to confirm the settings are correct.

On generating the code, I observe still Keil version 5 compatible / compilable port.c file and the folder RVDS is in use (instead of GCC).

Please help me with your inputs.

Thanks,
Rajeev

* External Message, Be Cautious *

aggarg, 2020-09-18:
> Is there a way to tell STM32CubeMX to use GCC and not use RVDS? 

STM32CubeMX uses RVDS port files and I do not know a way of telling it to use GCC port files instead - This is probably a question for ST.

> If I manually make a change to code, I am concerned that the next time I generate code with help of same IOC file, it may delete the GCC folder and instead add the RVDS folder.

The best I can suggest here is to use `git diff` (assuming you are using git) to examine and exclude RVDS changes from the auto generated code.

Thanks.

rxa1031, 2020-09-21:
Below information is shared as reference for individuals / coders who are looking out for answer to my query #Is there a way to tell STM32CubeMX to use GCC and not use RVDS?

I've raised concern about this to ST Community:

How to instruct STM32CubeMX to generate code compilable with ARM / Keil Compiler Version 6
Case Number: 00115320
Date/Time Opened: 9/18/2020 9:13 AM

Thanks,
@rxa1031

aggarg, 2020-09-23:
Given that ST has agreed to fix this in CubeMX, I am closing this issue. Feel free to re-open or create a new issue, if you need anything else.

Thanks.

closed 2020-09-23 by aggarg