FreeRTOS/FreeRTOS-Kernel #159: [BUG] configTASK_RETURN_ADDRESS is not respected

amirgon, 2020-09-07:
**Describe the bug**

`configTASK_RETURN_ADDRESS` is not set as the return address of task.

Looking at the code:

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/700c1cf9c630e19e9cd683c661dc22701fad2b2a/portable/GCC/RISC-V/port.c#L55-L59

However, neither `configTASK_RETURN_ADDRESS` nor `portTASK_RETURN_ADDRESS` are written to the task's stack.

**Target**

GCC/RISC-V

**Host**

Any

**To Reproduce**

- Set `configTASK_RETURN_ADDRESS` in FreeRTOSConfig.h to function `F`
- Create a task that returns
- Function `F` is not called

**Expected behavior**

`configTASK_RETURN_ADDRESS` should set the return address of a task that returns.

kstribrnAmzn, 2022-01-04:
Thank you for reaching out with this bug! You're correct that this is a problem. This is something we are aware of and are aiming to fix soon.

I'll try to keep you better updated when I have a concrete ETA.

aggarg, 2022-03-01:
Fixed in this PR - https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/444

closed 2022-03-01 by aggarg