FreeRTOS/FreeRTOS-Kernel #164: [BUG]

jeremyerdmann, 2020-09-10:
**Describe the bug**
xTimerCreate, xTimerCreateStatic, and vTimerSetReloadMode all take an argument const UBaseType_t uxAutoReload and uxTimerGetReloadMode returns UBaseType_t, yet the API documentation indicates valid values of pdTRUE and pdFALSE (which are defined as type BaseType_t in projdefs.h).  This incompatibility is flagged as a type mismatch with a MISRA-C checker (and possibly other static checkers) and requires a cast to UBaseType_t to resolve.  I'll note that this is being reported based on a "distribution" from STMicroelectronics, so it's possible that header file is from them, not part of the entire distribution.

**Target**
- Development board: NUCLEO-F429ZI
- Instruction Set Architecture: Cortex-M4
- IDE and version: IAR Embedded Workbench 8.42.2.24293
- Toolchain and version: IAR Embedded Workbench 8.42.2.24293

**Host**
- Host OS: MS Windows 10 Pro
- Version: 1909 (18363.900)

**To Reproduce**
- Compiling the following w/ MISRA-C checks enabled
```
        m_hFreeRTOSTimer = xTimerCreateStatic(
            name,
            pdMS_TO_TICKS(m_period_ms),
            isRepeating ? pdTRUE : pdFALSE, // Converts boolean isRepeating to either pdTRUE or pdFALSE
            this,
            TimerExpiredCallback,
            &m_timerControlBlock);
        ASSERT(m_hFreeRTOSTimer != NULL);
```
causes the following MISRA-C(2004) error: Error[Pm128]: illegal implicit conversion from underlying MISRA type "long" to "unsigned long" (MISRA C 2004 rule 10.1) based on the uxAutoReload being of type BaseType_t, yet the function signatures take a UBaseType_t.

**Expected behavior**
While I think I've read that FreeRTOS doesn't claim compliance with MISRA-C, this seems like a legitimate issue.

**Screenshots**
N/A

**Additional context**
N/A

jeremyerdmann, 2020-09-10:
File is not an ST file: https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/include/projdefs.h

syntroniks, 2020-09-16:
@jeremyerdmann Do things look better if you change `pdTRUE` and `pdFALSE` to `UBaseType_t`, or does that create more issues with your checker?

Either the timer function signatures would be changed, or projdefs would be changed. I personally wouldn't mind if pdTRUE and pdFALSE were unsigned.

jeremyerdmann, 2020-09-22:
Sorry for the late reply, I should really check this folder more often :(.

That pretty much just moves the issue.  A lot of functions returning bools
return a BaseType_t value
(e.g. xTimerCreateTimerTask, xTimerGenericCommand, and
xTimerIsTimerActive).  I think the issue is one of consistency.  They can
be one or the other, but not mix.  While unsigned might make more sense,
that might become a deeper rooted change.  The attached Git diff seemed to
get rid of the checker warning for me (note: I didn't touch commented out
code).  It did trickle into a number of calls (all changing the
uxAutoReload argument from UBaseType to BaseType):

xTimerCreate
xTimerCreateStatic
vTimerSetReloadMode
prvInitialiseNewTimer

If you go the other way, I'm guessing you'd have a tone of API return types
that you'd have to tweak throughout the code and documentation.

diff --git a/Middlewares/Third_Party/FreeRTOS/Source/include/timers.h b/Middlewares/Third_Party/FreeRTOS/Source/include/timers.h
index cb721797..e0833c98 100644
--- a/Middlewares/Third_Party/FreeRTOS/Source/include/timers.h
+++ b/Middlewares/Third_Party/FreeRTOS/Source/include/timers.h
@@ -227,7 +227,7 @@ typedef void (*PendedFunction_t)( void *, uint32_t );
 #if( configSUPPORT_DYNAMIC_ALLOCATION == 1 )
 	TimerHandle_t xTimerCreate(	const char * const pcTimerName,			/*lint !e971 Unqualified char types are allowed for strings and single characters only. */
 								const TickType_t xTimerPeriodInTicks,
-								const UBaseType_t uxAutoReload,
+								const BaseType_t uxAutoReload,
 								void * const pvTimerID,
 								TimerCallbackFunction_t pxCallbackFunction ) PRIVILEGED_FUNCTION;
 #endif
@@ -357,7 +357,7 @@ typedef void (*PendedFunction_t)( void *, uint32_t );
 #if( configSUPPORT_STATIC_ALLOCATION == 1 )
 	TimerHandle_t xTimerCreateStatic(	const char * const pcTimerName,			/*lint !e971 Unqualified char types are allowed for strings and single characters only. */
 										const TickType_t xTimerPeriodInTicks,
-										const UBaseType_t uxAutoReload,
+										const BaseType_t uxAutoReload,
 										void * const pvTimerID,
 										TimerCallbackFunction_t pxCallbackFunction,
 										StaticTimer_t *pxTimerBuffer ) PRIVILEGED_FUNCTION;
@@ -1246,7 +1246,7 @@ const char * pcTimerGetName( TimerHandle_t xTimer ) PRIVILEGED_FUNCTION; /*lint
  * uxAutoReload is set to pdFALSE then the timer will be a one-shot timer and
  * enter the dormant state after it expires.
  */
-void vTimerSetReloadMode( TimerHandle_t xTimer, const UBaseType_t uxAutoReload ) PRIVILEGED_FUNCTION;
+void vTimerSetReloadMode( TimerHandle_t xTimer, const BaseType_t uxAutoReload ) PRIVILEGED_FUNCTION;
 
 /**
  * TickType_t xTimerGetPeriod( TimerHandle_t xTimer );
diff --git a/Middlewares/Third_Party/FreeRTOS/Source/timers.c b/Middlewares/Third_Party/FreeRTOS/Source/timers.c
index 59b3840d..31ff661f 100644
--- a/Middlewares/Third_Party/FreeRTOS/Source/timers.c
+++ b/Middlewares/Third_Party/FreeRTOS/Source/timers.c
@@ -218,7 +218,7 @@ static void prvProcessTimerOrBlockTask( const TickType_t xNextExpireTime, BaseTy
  */
 static void prvInitialiseNewTimer(	const char * const pcTimerName,			/*lint !e971 Unqualified char types are allowed for strings and single characters only. */
 									const TickType_t xTimerPeriodInTicks,
-									const UBaseType_t uxAutoReload,
+									const BaseType_t uxAutoReload,
 									void * const pvTimerID,
 									TimerCallbackFunction_t pxCallbackFunction,
 									Timer_t *pxNewTimer ) PRIVILEGED_FUNCTION;
@@ -281,7 +281,7 @@ BaseType_t xReturn = pdFAIL;
 
 	TimerHandle_t xTimerCreate(	const char * const pcTimerName,			/*lint !e971 Unqualified char types are allowed for strings and single characters only. */
 								const TickType_t xTimerPeriodInTicks,
-								const UBaseType_t uxAutoReload,
+								const BaseType_t uxAutoReload,
 								void * const pvTimerID,
 								TimerCallbackFunction_t pxCallbackFunction )
 	{
@@ -308,7 +308,7 @@ BaseType_t xReturn = pdFAIL;
 
 	TimerHandle_t xTimerCreateStatic(	const char * const pcTimerName,		/*lint !e971 Unqualified char types are allowed for strings and single characters only. */
 										const TickType_t xTimerPeriodInTicks,
-										const UBaseType_t uxAutoReload,
+										const BaseType_t uxAutoReload,
 										void * const pvTimerID,
 										TimerCallbackFunction_t pxCallbackFunction,
 										StaticTimer_t *pxTimerBuffer )
@@ -348,7 +348,7 @@ BaseType_t xReturn = pdFAIL;
 
 static void prvInitialiseNewTimer(	const char * const pcTimerName,			/*lint !e971 Unqualified char types are allowed for strings and single characters only. */
 									const TickType_t xTimerPeriodInTicks,
-									const UBaseType_t uxAutoReload,
+									const BaseType_t uxAutoReload,
 									void * const pvTimerID,
 									TimerCallbackFunction_t pxCallbackFunction,
 									Timer_t *pxNewTimer )
@@ -439,7 +439,7 @@ Timer_t *pxTimer = xTimer;
 }
 /*-----------------------------------------------------------*/
 
-void vTimerSetReloadMode( TimerHandle_t xTimer, const UBaseType_t uxAutoReload )
+void vTimerSetReloadMode( TimerHandle_t xTimer, const BaseType_t uxAutoReload )
 {
 Timer_t * pxTimer =  xTimer;

RichardBarry, 2020-09-22:
Is my understanding that the main issue here is that the 'auto reload' value is unsigned, and making it signed would keep the static checkers happy?

jeremyerdmann, 2020-09-22:
yes, that is correct