FreeRTOS/FreeRTOS-Kernel #204: [BUG] incorrect implementation of pvPortMalloc

TheSilentDawn, 2020-10-20:
**Describe the bug**

The function pvPortMalloc() is used to allocate memory for the required size.
The parameter xWantedSize indicates the length of memory required. However, when the required memory block size is zero. The memory pointer variable pvReturn will keep as NULL. If the configuration configUSE_MALLOC_FAILED_HOOK is enabled, it will recognize the variable xWantedSize valued zero as an error and handle it with the function vApplicationMallocFailedHook() as shown in https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/5fb26de0194b4ebfd09b37128bf9f3ff6d34c1b7/portable/MemMang/heap_4.c#L253

**Target**
- Development board: STM32H7B3I-DK and STM32H7B3I-EVAL
- Instruction Set Architecture: Thumb
- IDE and version: STMCubeIDE 1.4.0
- Toolchain and version: Default toolchain of STMCubeIDE

**Host**
- Host OS: Ubuntu
- Version: 18.04 LTS

**To Reproduce**
- STM32H7 MCU package. STM32H7B3I-DK/Domonstrations/ClockAndWeather. Default configuration for debug.
- Run the example and the firmware loops infinitely. 

**Expected behavior**

Normally, if the variable xWantedSize is zero, a pointer should be returned successfully, no matter the NULL pointer or a real pointer with the next available heap address.
As a result, the firmware could continue to execution.

RichardBarry, 2020-10-20:
Thanks for taking the time to write this up - but I do not believe this is a bug.  According to ANSI C:  "If the size of the space requested is 0, the behavior is implementation-defined: the value returned shall be either a null pointer or a unique pointer."

TheSilentDawn, 2020-10-20:
Hello,

Thanks for your reply. Based on the ANSI C, the value returned shall be either a null pointer or a unique pointer. However, the heap_4.c doesn’t follow this guide. It will not return but hang into the function vApplicationMallocFailedHook() when configUSE_MALLOC_FAILED_HOOK is enabled. Based on my understanding, vApplicationMallocFailedHook()  should be called when the left memory is not enough for the required size, not the zero size. This bug will actually stop some firmware, for example, the ClockAndWeather example of STM32H7B3I-EVAL and STM32H7B3I-DK.

TheSilentDawn, 2020-10-20:
@RichardBarry Thanks for your reply. Based on the ANSI C, the value returned shall be either a null pointer or a unique pointer. However, the heap_4.c doesn’t follow this guide. It will **return neither** but hang into the function vApplicationMallocFailedHook() when configUSE_MALLOC_FAILED_HOOK is enabled. Based on my understanding, vApplicationMallocFailedHook()  should be called when the left memory is not enough for the required size, not the zero size. This bug will actually stop some firmware, for example, the ClockAndWeather example of STM32H7B3I-EVAL and STM32H7B3I-DK.

RichardBarry, 2020-10-20:
vApplicationMallocFailedHook() getting called is not a fatal error.  It is perfectly legitimate for allocations to fail at run time, and production code (although not a lot of our demos) should recover from that.

I just tried updating the test to used to determin if the allocation fail hook is called to:
```
if( ( pvReturn == NULL ) && ( xWantedSize != ( size_t) 0 ) )
```

...and that seems to work.  The change is probably needed in all the heap_n.c files.

Grateful if you can create a pull request in GIthub with this change.

TheSilentDawn, 2020-10-20:
> vApplicationMallocFailedHook() getting called is not a fatal error. It is perfectly legitimate for allocations to fail at run time, and production code (although not a lot of our demos) should [...]

TheSilentDawn, 2020-10-20:
The solution of the function vApplicationMallocFailedHook() seems to work.
Yes, other heap_n.c also have the same problem. 
Thanks.

TheSilentDawn, 2020-10-20:
The solution of the function vApplicationMallocFailedHook() seems to work.
Yes, other heap_n.c also has the same problem. 
Thanks.

TheSilentDawn, 2020-10-21:
@RichardBarry Hello, I rechecked the patch or advise implementation of the function vApplicationMallocFailedHook(). 
`if( ( pvReturn == NULL ) && ( xWantedSize != ( size_t) 0 ) )`
This is helpless because the function vApplicationMallocFailedHook() has no parameters. As a result, it couldn't check the variable *pvReturn* and *xWantedSize*. FreeRTOS may require the third-party vendor to implement this function by the macro definition or modify this function by adding parameters, but all these are friendless. As a result, based on ASNI C advisement, I still think FreeRTOS should be responsible for returning NULL or a valid pointer when xWantedSize is zero and the function vApplicationMallocFailedHook() only in charge of failing to allocate a new memory block.

n9wxu, 2021-12-27:
malloc(0) responses are implementation defined.

closed 2021-12-27 by n9wxu