FreeRTOS/FreeRTOS-Kernel #211: Git tag schema is confusing

Viatorus, 2020-10-27:
**Describe the bug**

There exists two types of git tags:
`V10.4.1-kernel-only` and `V10.3.0`

And both have different folder and file structure... If someone is using a package manager (like conan) a clone and build from the newest kernel version isn't that easy.

So that we can rely on: Will all future releases of the kernel be tagged `<version>-kernel-only` or could there be a `V10.5.0` (including the kernel update) without a `V10.5.0-kernel-only` tag?

cobusve, 2020-10-29:
Hi @Viatorus,

We are using semantic versioning as defined here https://semver.org/

As you can see on that page dashes are part of the scheme such that (I quote from semver.org):
  Example: 1.0.0-alpha < 1.0.0-alpha.1 < 1.0.0-alpha.beta < 1.0.0-beta < 1.0.0-beta.2 < 1.0.0-beta.11 < 1.0.0-rc.1 < 1.0.0.

I have cheked the conan documentation and it claims to by default follow semver. (see https://docs.conan.io/en/latest/versioning/introduction.html)

From your report it sounds like you had a specific problem. Perhaps you can give a little more detail so that we can understand exactly what did not work for you?

Viatorus, 2020-10-29:
Hello @cobusve,

thank you for your explanation of semantic versioning, but I am already aware of that. This wasn't my question. I try to explain my intention/question again:

Have a look at all tags: https://github.com/FreeRTOS/FreeRTOS-Kernel/tags

Starting with V10.3.0 this repository (FreeRTOS-Kernel) has a new tag schema: V10.3.0-kernel, which indicates updates of the kernel, without FreeRTOS-Plus and tools. Will the other tag schema (without _-kernel_) will be continued for this repository (like in https://github.com/FreeRTOS/FreeRTOS) or dropped? Since there isn't a tag for V10.4.1 but only V10.4.1-kernel here.

If so, I would always fetch the FreeRTOS-Kernel from this repository from tags like vXX.XX.XX-kernel from now on. Previous I used the tag without _-kernel_, because the other didn't exists.

cobusve, 2020-10-30:
Ok I think I understand. Let me explain what is going on here.

The short answer to your question is that this repository no longer contains the FreeRTOS-Plus or Labs files. If we decide to make a version V10.5.0 it will have the same folder structure as V10.4.1-Kernel-only and will contain only the Kernel. This is reflected in the description of this repo which reads as follows:
 _"FreeRTOS kernel files only, submoduled into https://github.com/FreeRTOS/FreeRTOS and various other repos."_

I think frankly what you are looking for is most likely the hub repository in https://github.com/FreeRTOS/FreeRTOS as that contains the other libraries (excluding the Labs code). The description for that repository reads as follows:
_"'Classic' FreeRTOS distribution. Started as Git clone of FreeRTOS SourceForge SVN repo. Submodules the kernel."_

Of course historically (in SVN) all the code was together so when we tagged older versions here we wanted to make sure the result matched what was originally released from SVN. Because of this versions that were created before moving to GitHub still contain the older files.

Further to this - we have moved many of the FreeRTOS-Plus libraries such as FreeRTOS-Plus-TCP to their own repositories to make it possible to consume these into your projects one at a time. For example the Plus-TCP code now lives here https://github.com/FreeRTOS/FreeRTOS-Plus-TCP and is pulled into the hub repository as a git submodule.

Viatorus, 2020-10-30:
Thank you for your explanation.

But I think I stay with this repository and the _-kernel-only_ thing, because we just need the kernel. No Plus, no demos or tests.

Thank you!

cobusve, 2020-10-30:
That is perfect, this repository was created exactly for people who need only the Kernel.

Viatorus, 2020-11-11:
So now where is the V10.4.2-Kernel-only tag? :)

cobusve, 2020-11-12:
I explained this above when I said this 

"The short answer to your question is that this GitHub repository no longer contains the FreeRTOS-Plus or Labs files. It really never did because this GitHub repository was created after V10.3.0 was released. If we do decide to make a version V10.5.0 it will have the same folder structure as V10.4.1-Kernel-only and will contain only the Kernel. This is reflected in the description of this repo which reads as follows:"

If you want to use only the Kernel you should use the repository which is called https://github.com/FreeRTOS/FreeRTOS-Kernel. All releases in this repo will always only contain the kernel (with the exception of releases that were made before this repository existed).

What you need to use is the latest semantic version in this repository if you want only the kernel.

closed 2020-11-12 by cobusve