FreeRTOS/FreeRTOS-Kernel #234: Please add gitattributes file - mixed line endings make repo hard to use

tnh8, 2020-12-17:
**Describe the issue**
Having mixed line endings and not using .gitattributes is making repository harder to use. 

**Reference**
https://github.com/FreeRTOS/FreeRTOS-Kernel

**Screenshot**
`find . -not -type d -exec file "{}" ";"`

[image]

dachalco, 2020-12-23:
Hi @tnh8

Good suggestion. I've Internally queued an action item to investigate this. Someone should be following up with a resolution soon. Thanks.

dachalco, 2021-01-15:
Hi @tnh8 

Thanks again for the suggestion. I've added a `.gitattributes` to both [FreeRTOS/FreeRTOS](https://github.com/FreeRTOS/FreeRTOS/pull/490) and [FreeRTOS/FreeRTOS-Kernel](https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/245).

closed 2021-01-15 by dachalco