FreeRTOS/FreeRTOS-Kernel #247: [BUG] Recently added .gitattributes converts line endings to CRLF on Linux

compihu, 2021-01-17:
**Describe the bug**
After cloning the repository all files are modified silently bu git converting LF to CRLF

'''
xxx@yyy:/mnt/userdata/xxx/work/ext$ git clone git@github.com:FreeRTOS/FreeRTOS-Kernel.git
Cloning into 'FreeRTOS-Kernel'...
remote: Enumerating objects: 4, done.
remote: Counting objects: 100% (4/4), done.
remote: Compressing objects: 100% (4/4), done.
remote: Total 151604 (delta 0), reused 0 (delta 0), pack-reused 151600
Receiving objects: 100% (151604/151604), 104.39 MiB | 8.30 MiB/s, done.
Resolving deltas: 100% (109604/109604), done.
cd FreeRTOS-Kernel
xxx@yyy:/mnt/userdata/xxx/work/ext/FreeRTOS-Kernel$ git status
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   .github/CONTRIBUTING.md
        modified:   .github/SECURITY.md
        modified:   croutine.c
        modified:   event_groups.c
        modified:   include/FreeRTOS.h
        modified:   include/StackMacros.h
        modified:   include/atomic.h
        modified:   include/croutine.h
        modified:   include/deprecated_definitions.h
        modified:   include/list.h
        modified:   include/message_buffer.h
        modified:   include/mpu_prototypes.h
        modified:   include/mpu_wrappers.h
        modified:   include/portable.h
        modified:   include/projdefs.h
        modified:   include/semphr.h
        modified:   include/stack_macros.h
        modified:   include/stream_buffer.h
        modified:   list.c
        modified:   portable/ARMv8M/copy_files.py
...
'''

xgroleau, 2021-01-18:
We have the same issue, it created issues on our CI since we pull freertos as a dependency.

podhrmic, 2021-01-19:
Same here. Changing a line in a file results in whole file being changed, see for example #249

dachalco, 2021-01-27:
Yikes yes it is very inconvenient. Deleted the file and planning to defer line-ending unification to our code beautifier.

closed 2021-01-27 by dachalco