FreeRTOS/FreeRTOS-Kernel #252: [BUG] FreeRTOS 10.2.1 fails to build in IAR EWARM in C++ mode

BFlemingFDI, 2021-01-22:
**Describe the bug**
FreeRTOS 10.2.1 fails to build in IAR EWARM in C++ mode
Note that this is the FreeRTOS version provided by the STM development tools code generator, but part of this bug may be present in a newer FreeRTOS release.

**Target**
- Development board: STM32L5
- Instruction Set Architecture: Cortex-M33
- IDE and version: IAR EWARM 8.50.9
- Toolchain and version: IAR Compiler

**Host**
- Host OS: Windows 

**To Reproduce**
- Use STM32CubeMX to generate an IAR project with FreeRTOS for a Cortex-M MCU. (STM32F7, STM32L5, etc)
- Open the project in IAR EWARM 8.50.
- Open the project properties of the project.
- Switch project to C++ mode. (see screenshot).
- Save project changes.
- Attempt to build. (requires IAR license activation)

**Screenshots**
Build Failure when switching to C++ Mode:
[image]

**Proposed Fix**
Here is a patch to fix the issue. Just need to cast everything correctly.
The software still builds on GCC after the patch.
[IAR_FreeRTOS_Patch.diff.txt](https://github.com/FreeRTOS/FreeRTOS-Kernel/files/5858236/IAR_FreeRTOS_Patch.diff.txt)

RichardBarry, 2021-01-22:
Thanks - we will take a look and update as necessary - and also check the impact on the static checking results.  Wanted to note that cmsis files are not ours though, by which I mean we didn't author them and don't distribute them, so you may want to let ST (or ARM if ARM were the original author) know too.

kevinpt, 2021-11-08:
The FreeRTOS and CMSIS source shouldn't be built in C++ mode. C++ doesn't allow automatic casts to/from void * which is idomatic for C. You can configure IAR to use the Auto language setting based on the file extension. Another option is to put all FreeRTOS and other C code in a group and configure that group to build with C as the language.

closed 2021-12-27 by n9wxu