FreeRTOS/FreeRTOS-Kernel #257: [BUG] Wrong type in definition of  vApplicationStackOverflowHook

DavidJurajdaNXP, 2021-02-05:
**Describe the bug**
On the link https://www.freertos.org/Hardware-independent-RTOS-example.html is presented generic example, but compilation fail with error due the conflicting types for 'vApplicationStackOverflowHook'.

Definition in example:
void vApplicationStackOverflowHook(TaskHandle_t xTask, signed char *pcTaskName)

Declaration in freertos_kernel/include/task.h:1616:11
void vApplicationStackOverflowHook( TaskHandle_t xTask, char * pcTaskName );

In FreeRTOS Kernel V10.3.0 has been function vApplicationStackOverflowHook referenced in task.c file via extern declaration. Although extern reference types are not alligned with function definition it does not result in error.

In FreeRTOS Kernel V10.4.3 is function vApplicationStackOverflowHook referenced in task.c via interface header file task.h. Misalignment between function declaration and definition result in error.

**Target**
- IDE and version: armgcc, Keil MDK, IAR EWARM

**Host**
- Windows 10, Linux, Mac

**Expected behavior**
Hardware independent FreeRTOS example from link above is buildable with recent version of FreeRTOS kernel.

RichardBarry, 2021-02-05:
Sorry but its not clear if you are reporting a bug in the code or that code cut and paste from the website doesn't build.  I'm assuming the latter, so have removed the "singed char" qualifiers from the html code and will close this ticket.  If you are instead reporting a bug in the code then please reopen the ticket and provide a link to the offending code in Github - thanks.

closed 2021-02-05 by RichardBarry

DavidJurajdaNXP, 2021-02-08:
Thank you @RichardBarry, fix in web example is appropriate. Code in repository is correct.