FreeRTOS/FreeRTOS-Kernel #261: [Feature Request] Enhhanced mutex priority inheritance algorithm

rhajdaj, 2021-02-11:
**Is your feature request related to a problem? Please describe.**
See this thread for a good, detailed explanation of the desired enhancement (esp, the initial entry by mastupristi):
https://forums.freertos.org/t/priority-inheritance-proposal/11175
See attached image.

**Describe the solution you'd like**
An enhanced version of the priority inversion algorithm that disinherits the promoted task when the mutex that caused the inheritance is given back (as opposed to the last mutex held by the task is given back).

**Describe alternatives you've considered**
1. A mutex semaphore, in which case the developer must be sure that no other mutexes are taken while the outer mutex is held.  This takes additional effort (esp with third party software) by developers who want to be "lazy" with regard to mutex use.
2. A binary semaphore, which is subject to priority inversion.
We're currently using option 2, but would prefer to use a mutex to protect against priority inversion.

**Additional context**
For Xilinx users, this issue is particularly problematic because many Xilinx drivers use "busy waits" for I/O out of the box.  Because of this, if a task that calls "busy CPU" driver functions is disinherited later than it could be, this will mean 100% starvation for all tasks below or equal to the inherited priority until the loop finishes, which can be a long time for large I/O operations.
[image]

svenbieg, 2024-08-31:
I'm not using prioritization at all. If there is something time critical, like my heap, i'm marking the task as exclusive, preventing interrupts. For this purpose i have implemented a task-lock, You can call it however You like. This guarantees the task to get 100% priority for the time of accessing the heap. This is my concept of prioritization.

svenbieg, 2024-09-02:
With my solution it is just impossible to use the heap from within an interrupt service routine, i'd use a condition_variable at this point.

n9wxu, 2024-09-03:
The thread linked in the description has an extensive discussion on the costs of different semaphore options.  Currently the the simplified PI algorithm used by FreeRTOS is sufficient for most applications.

We would welcome a PR that accomplished the following:
1. extracted the semaphore into a separate file to simplify replacement.
2. An implementation of an alternative with performance measurements.

The discussion of high performance semaphores and the alternatives is very popular and deserves continual discussion.  However we have no plans to change the current implementation at this time.

closed 2024-09-03 by n9wxu