FreeRTOS/FreeRTOS-Kernel #297: [Feature Request] Can we get rid of xxxFromISR() functions or catch wrong function call (nonFromISR function call in ISR)?

Hadatko, 2021-03-30:
**Is your feature request related to a problem? Please describe.**
We found out after months that some function which is called in ISR was missing fromISR postfix. But application start rebooting recently. 

**Describe the solution you'd like**
For example RPMSG contains platform and environment function to figure out if the call is inside or outside of ISR and based on that they are calling rtos API. It would be nice if rtos itself could handle that. Maybe have some function to implement from users. I saw also xPortIsInsideInterrupt. So after that change freertos would have single api for in ISR and outside of ISR calls. 

**Describe alternatives you've considered**
Other possible solution could be inform somehow user that he is calling wrong function in/outside ISR.

**Additional context**
-

RichardBarry, 2021-03-30:
Thanks for your feedback.  There is an [FAQ](https://www.freertos.org/FAQ_API.html#IQRAPI) item here as to why the code is as it is - historical decisions.  The 'FromISR' versions are often used as simpler/faster versions for use within tasks too - although some care needs to be done when doing that.  Perhaps a quick thing to do would be to add asserts into the taskENTER_CRITICAL() and taskEXIT_CRITICAL() macros that triggers if the macros are called from an ISR.

Hadatko, 2021-04-01:
I understand that it can be harder to implement both feature with one API name. Your suggestion sounds good to me. And can be som check added also for oposite situation?

john-mullan-agilent, 2022-08-24:
I ran into similar issues with FreeRTOS API functions and had luck writing my own API wrappers with a check for the condition (portNVIC_INT_CTRL_REG & portVECTACTIVE_MASK) to call the appropriate FreeRTOS API's as needed. This requires copying the portVECTACTIVE_MASK definition from port.c into user code, and maybe a FreeRTOS API could be provided to check this condition.

closed 2023-03-29 by chinglee-iot