FreeRTOS/FreeRTOS-Kernel #311: [BUG]

giso-c, 2021-04-15:
in freertos V10.4.3 in queue.c, an extra ConfigASSERT as been added for pcQueuName.

    void vQueueAddToRegistry( QueueHandle_t xQueue,
                              const char * pcQueueName ) /*lint !e971 Unqualified char types are allowed for strings and single characters only. */
    {
        UBaseType_t ux;

        configASSERT( xQueue );
        configASSERT( pcQueueName );

In previous version the name could be NULL, Is it a necessary to have pcQueueName  defined to non NULL.
My code was running fine before with pcQueueName  NULL, but now I get this assert.

AniruddhaKanhere, 2021-04-15:
Hello @giso-c,

Yes, you are right. This assertion was not there earlier.
I am not an expert but what I think is that a NULL value in the pcQueueName is used to show that the location is unused.
See this code: https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/queue.c#L2814-L2815.

paulbartell, 2021-04-15:
@giso-c : Could you elaborate on your use case a bit? Are you using a particular rtos-aware debugger?

giso-c, 2021-04-16:
I am using IAR compiler, HW paltform is STM32 ARM chip.
I also have CMSIS2 interface. in my code I just create a queue
osMessageQueueNew(1, sizeof(uint32_t), NULL);

in cmsis_os2.c 
osMessageQueueId_t osMessageQueueNew (uint32_t msg_count, uint32_t msg_size, const osMessageQueueAttr_t *attr) {
  QueueHandle_t hQueue;
  int32_t mem;
  #if (configQUEUE_REGISTRY_SIZE > 0)
  const char *name;
  #endif

  hQueue = NULL;

  if ((IRQ_Context() == 0U) && (msg_count > 0U) && (msg_size > 0U)) {
    mem = -1;

    if (attr != NULL) {
      if ((attr->cb_mem != NULL) && (attr->cb_size >= sizeof(StaticQueue_t)) &&
          (attr->mq_mem != NULL) && (attr->mq_size >= (msg_count * msg_size))) {
        mem = 1;
      }
      else {
        if ((attr->cb_mem == NULL) && (attr->cb_size == 0U) &&
            (attr->mq_mem == NULL) && (attr->mq_size == 0U)) {
          mem = 0;
        }
      }
    }
    else {
      mem = 0;
    }

    if (mem == 1) {
      hQueue = xQueueCreateStatic (msg_count, msg_size, attr->mq_mem, attr->cb_mem);
    }
    else {
      if (mem == 0) {
        hQueue = xQueueCreate (msg_count, msg_size);
      }
    }

    #if (configQUEUE_REGISTRY_SIZE > 0)
    if (hQueue != NULL) {
      if (attr != NULL) {
        name = attr->name;
      } else {
        name = NULL;
      }
      vQueueAddToRegistry (hQueue, name);
    }
    #endif
  }
  return ((osMessageQueueId_t)hQueue);
}

I had a closer look at the code and effectivly NULL means empty slot. 
When creating semaphore, queue, mutex, cmsis layer is calling vQueueAddToRegistry even when name == NULL.
The new assert in queue.c is just pointing out an issue of cmsis code.

PR #313 (2021-04-17): Replace configASSERT( pcQueueName ) in vQueueAddToRegistry with a NULL pointer check.
Replace configASSERT( pcQueueName ) in vQueueAddToRegistry with a NULL pointer check.

Description
-----------
Replace the configASSERT added to vQueueAddToRegistry with a NULL pointer check so that it does not break usage in https://github.com/ARM-software/CMSIS-FreeRTOS.

Test Steps
-----------
Unit test currently in a Pull Request will be updated.

Related Issue
-----------
Fixes #311

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the t [... cut]
closed 2021-04-17 by RichardBarry