FreeRTOS/FreeRTOS-Kernel #312: [Feature Request] Enable each stream buffer to use a different sbSEND_COMPLETED implementation

RichardBarry, 2021-04-15:
**Is your feature request related to a problem? Please describe.**
sbSEND_COMPLETED is the macro that performs any processing necessary after data has been sent to a stream buffer.  By default this will unblock a task if a task is waiting for data from the stream buffer.  The application writer can provide their own definition that, for example, can raise an interrupt in another core when the stream buffer is used for core to core communication.  However currently a single global definition of sbSEND_COMPLETED applies to every stream buffer, enforcing each stream buffer to behave in the same way.  

**Describe the solution you'd like**
While the above can be worked around within the application it would be an improvement if each stream buffer can define its own sbSEND_COMPLETED equivalent, and only use the default implementation when  a stream buffer specific implementation is not supplied.

**Additional context**
See https://forums.freertos.org/t/messagebuffers-for-core-to-core-communication/12318

eddie9712, 2021-06-28:
Hi @RichardBarry , I've sent a PR related to this issue would you please check if the solution is that you would like.

RichardBarry, 2021-06-28:
Yes - thanks - I've checked out the branch already but git is telling me I have local changes in my repo (which I don't) so so far I've only done a visual diff.

eddie9712, 2021-06-28:
Ok, If there is anything I can help please let me know :)

kar-rahul-aws, 2023-09-28:
Closing this request, since PR #437 which addresses this request is merged.

closed 2023-09-28 by kar-rahul-aws