FreeRTOS/FreeRTOS-Kernel #316: [Feature Request] <Support Cortex-A35>

cyliangtw, 2021-04-23:
**Is your feature request related to a problem? Please describe.**
So far, the support Cortex-A CPU cores are A9 and A53. I can't find CPU core Cortex-A35 64-bit in the list.

**Describe the solution you'd like**
I am eager to see Cortex-A35 64-bit in [GCC CPU list](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/GCC) or [RVDS CPU list](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/RVDS) ,then I could apply FreeRTOS on our Cortex-A35 platform.

**Describe alternatives you've considered**
None

**Additional context**
None

RichardBarry, 2021-05-21:
What are the major differences between the A35 and A55 (64-bit)?

cyliangtw, 2021-05-21:
This web page [arm-cortex-a35-in-comparison](https://www.toradex.com/blog/the-arm-cortex-a35-in-comparison) comparing the Cortex-A35 and Cortex-A53. 
In RTOS porting layer, it's relative to ARM CPU instructions, for example Cortex-M3 is different from Cortex-M4 in APSR, xPSR, SCB at least.

Atomar25, 2021-05-25:
I don't think, there is lot of difference Cortex-A35 and Cortex-A53 when it comes to CPU instructions, Cortex-A35 is Armv8-A 64-bit processor which is what Cortex-A53 based on. So, I believe any code that runs on Cortex-A53 should work on Cortex-A35. It just you need to figure out what the Memory Map your SBC provides and enable the Device (GIC and UART to start with).

I would like to try this out but unfortunately doesn't have an SBC with Cortex-A35.

cyliangtw, 2021-06-04:
@Atomar25 , Thanks of your comment, according to my experience in Cortex-M23, both of Cortex-M23 and Cortex-M33 are Armv8M, the CPU instructions are similar but still need to divide as ARM_CM23 and ARM_CM33 at [portable/GCC](https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/GCC)

If just verify CPU core only, keil Simulator is a good choice. In Keil IDE project options, press "Debug" and check radio button of "Use Simulator".

sean-ascent, 2024-09-03:
@RichardBarry @Atomar25  I would like to revive this request and add a few contributions from my side. I just finished making modifications to the AARCH64 port to get the kernel running on an A35 core. Details below:

**Target**
- Development board: **STM32MP257F-EV1**
- Instruction Set Architecture: **A64**
- IDE and version: **VsCode 1.92.2**
- Toolchain and version: **arm-gnu-toolchain-13.2.Rel1-x86_64-aarch64-none-elf**

**Host**
- Host OS: **Ubuntu (Docker)**
- Version:  **22.04**

**Expected Behavior**
Kernel will preemptively switch tasks via either the `FreeRTOS_SWI_Handler` (by way of SMC instruction) or via the `FreeRTOS_Tick_Handler` in conjunction with the `FreeRTOS_IRQ_Handler`.

**Actual Behavior**
Kernel successfully switches to initial task via `FreeRTOS_SWI_Handler` but subsequent switch attempts by the tick mechanism result in an ambiguous synchronous exception (ESR = 0x0, ELR = 0x3C).

**Modifications**

The STM32MP2xx microprocessor utilizes GIC v2.0 which supports interrupt grouping and security extensions. By  architecture, all non-secure IRQs are routed to the GIC as IRQs and all secure IRQs are routed as FIQs. The AARCH64 port currently does not correctly handle the discovery and clearing of IRQs by interrupt group. The AARCH64 port was modified to utilize the ICCHPPIR register, in conjunction with the aliased and non-aliased ICCIAR and ICCEOIR registers, to correctly handle interrupts for this part. Interrupt pending bit clearing is delegated to user `vApplicationIRQHandler`. Context switching occurs via `FreeRTOS_Tick_Handler` call from the physical timer interrupt handler.

**Diff** 

portable/GCC/ARM_AARCH64/port.c
```diff
diff --git a/portable/GCC/ARM_AARCH64/port.c b/portable/GCC/ARM_AARCH64/port.c
index 238874edc..7d2b3e673 100644
--- a/portable/GCC/ARM_AARCH64/port.c
+++ b/portable/GCC/ARM_AARCH64/port.c
@@ -166,9 +166,12 @@ uint64_t ullPortYieldRequired = pdFALSE;
 uint64_t ullPortInterruptNesting = 0;
 
 /* Used in the ASM code. */
-__attribute__( ( used ) ) const uint64_t ullICCEOIR = portICCEOIR_END_OF_INTERRUPT_REGISTER_ADDRESS;
-__attribute__( ( used ) ) const uint64_t ullICCIAR = portICCIAR_INTERRUPT_ACKNOWLEDGE_REGISTER_ADDRESS;
 __attribute__( ( used ) ) const uint64_t ullICCPMR = portICCPMR_PRIORITY_MASK_REGISTER_ADDRESS;
+__attribute__( ( used ) ) const uint64_t ullICCHPPIR = portICCHPPIR_HIGHEST_PRIORITY_INTERRUPT_REGISTER_ADDRESS;
+__attribute__( ( used ) ) const uint64_t ullICCIAR = portICCIAR_INTERRUPT_ACKNOWLEDGE_REGISTER_ADDRESS;
+__attribute__( ( used ) ) const uint64_t ullICCEOIR = portICCEOIR_END_OF_INTERRUPT_REGISTER_ADDRESS;
+__attribute__( ( used ) ) const uint64_t ullICCAIAR = portICCAIAR_ALIASED_INTERRUPT_ACKNOWLEDGE_REGISTER_ADDRESS;
+__attribute__( ( used ) ) const uint64_t ullICCAEOIR = portICCAEOIR_ALIASED_END_OF_INTERRUPT_REGISTER_ADDRESS;
 __attribute__( ( used ) ) const uint64_t ullMaxAPIPriorityMask = ( configMAX_API_CALL_INTERRUPT_PRIORITY << portPRIORITY_SHIFT );
 
 /*-----------------------------------------------------------*/
```

portable/GCC/ARM_AARCH64/portASM.S
``` diff
diff --git a/portable/GCC/ARM_AARCH64/portASM.S b/portable/GCC/ARM_AARCH64/portASM.S
index e684755bf..f75d34dd8 100644
--- a/portable/GCC/ARM_AARCH64/portASM.S
+++ b/portable/GCC/ARM_AARCH64/portASM.S
@@ -307,13 +307,32 @@ FreeRTOS_IRQ_Handler:
     /* Maintain the interrupt nesting information across the function call. */
     STP     X1, X5, [SP, #-0x10]!
 
-    /* Read value from the interrupt acknowledge register, which is stored in W0
-    for future parameter and interrupt clearing use. */
-    LDR     X2, ullICCIARConst
-    LDR     X3, [X2]
-    LDR     W0, [X3]    /* ICCIAR in W0 as parameter. */
-
-    /* Maintain the ICCIAR value across the function call. */
+    /* Read value from the HPPIR register 
+     * This determines which interrupt group the IRQ belongs to.
+     */
+    LDR     X2, ullICCHPPIRConst
+    LDR     X2, [X2]
+    LDR     W3, [X2]
[... 91 lines cut ...]
+#define portINTERRUPT_CONTROLLER_CPU_INTERFACE_ADDRESS              ( configINTERRUPT_CONTROLLER_BASE_ADDRE
SS + configINTERRUPT_CONTROLLER_CPU_INTERFACE_OFFSET )
+#define portICCPMR_PRIORITY_MASK_REGISTER                           ( *( ( volatile uint32_t * ) ( portINTE
RRUPT_CONTROLLER_CPU_INTERFACE_ADDRESS + portICCPMR_PRIORITY_MASK_OFFSET ) ) )
+#define portICCHPPIR_HIGHEST_PRIORITY_INTERRUPT_REGISTER_ADDRESS    ( portINTERRUPT_CONTROLLER_CPU_INTERFAC
E_ADDRESS + portICCHPPIR_HIGHEST_PRIORITY_INTERRUPT_OFFSET )
+#define portICCIAR_INTERRUPT_ACKNOWLEDGE_REGISTER_ADDRESS           ( portINTERRUPT_CONTROLLER_CPU_INTERFAC
E_ADDRESS + portICCIAR_INTERRUPT_ACKNOWLEDGE_OFFSET )
+#define portICCPMR_PRIORITY_MASK_REGISTER_ADDRESS                   ( portINTERRUPT_CONTROLLER_CPU_INTERFAC
E_ADDRESS + portICCPMR_PRIORITY_MASK_OFFSET )
+#define portICCEOIR_END_OF_INTERRUPT_REGISTER_ADDRESS               ( portINTERRUPT_CONTROLLER_CPU_INTERFAC
E_ADDRESS + portICCEOIR_END_OF_INTERRUPT_OFFSET )
+#define portICCAIAR_ALIASED_INTERRUPT_ACKNOWLEDGE_REGISTER_ADDRESS  ( portINTERRUPT_CONTROLLER_CPU_INTERFAC
E_ADDRESS + portICCAIAR_ALIASED_INTERRUPT_ACKNOWLEDGE_OFFSET )
+#define portICCAEOIR_ALIASED_END_OF_INTERRUPT_REGISTER_ADDRESS      ( portINTERRUPT_CONTROLLER_CPU_INTERFAC
E_ADDRESS + portICCAEOIR_ALIASED_END_OF_INTERRUPT_OFFSET )
+#define portICCBPR_BINARY_POINT_REGISTER                            ( *( ( const volatile uint32_t * ) ( po
rtINTERRUPT_CONTROLLER_CPU_INTERFACE_ADDRESS + portICCBPR_BINARY_POINT_OFFSET ) ) )
+#define portICCRPR_RUNNING_PRIORITY_REGISTER                        ( *( ( const volatile uint32_t * ) ( po
rtINTERRUPT_CONTROLLER_CPU_INTERFACE_ADDRESS + portICCRPR_RUNNING_PRIORITY_OFFSET ) ) )
 
 #define portMEMORY_BARRIER()    __asm volatile ( "" ::: "memory" )
```

Hope this helps.

n9wxu, 2024-09-03:
Can you submit a PR with your changes?