FreeRTOS/FreeRTOS-Kernel #339: [BUG] Many source files have mixed line endings

cmcqueen, 2021-05-31:
If you run Unix tool `file` on the source files, many will report eg

    ./include/atomic.h: C source, ASCII text, with CRLF, LF line terminators

That **with CRLF, LF line terminators** means the files have some line endings of the form CR/LF, and some LF only. What a mess!

All files should have consistent line endings, either all CR/LF, or all LF.

For git cross-platform usage, I find it's generally preferable to use LF.

RichardBarry, 2021-05-31:
Different port layers may have different line endings depending on their origin and target, but each file should be consistent within itself, so I would like to understand this better.  I cant see which line [in the file you call out](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/include/atomic.h) is using just LF - can you link to the offending line within that file (or just let me know the line number within that link) - thanks.

RichardBarry, 2021-05-31:
I think I see it now - it is where: " * SPDX-License-Identifier: MIT" has been added to the comment block at the top of the file headers.

cmcqueen, 2021-05-31:
I don't know of a tool to show it easily. I manually looked at one file (not `atomic.h`) with a hex editor, and saw also that it was the licence-identifier line that was to blame.

You could probably use `git blame` to work out who messed it up in the first place, and then inquire about what text editor they were using.

It's probably necessary to automate a fix with some sort of batch tool.

cmcqueen, 2021-06-01:
It looks like it is due mainly to commit 3a413d1022a4ff60ffc9631877992ac08affcc21.

cobusve, 2021-06-01:
Thanks for reporting this! That change was made using a script and it highlighted that we have no tool in place to catch when this happened. I asked pbartell to fix this and add a PR check so that this does not happen again.

paulbartell, 2021-06-01:
My apologies. My script did not take into account line endings when I made that change and my core.autocrlf setting was not set correctly to compensate.

I will discuss with @RichardBarry. Here's my plan to address this:
1. Update the .gitattributes file to mark files as text by default (so git can handle line endings)
2. (Possibly) enable text=auto for all files and renormalize all line endings in the repository to CRLF
3. Add a pre-commit configuration file to allow developers to check for mixed line endings prior to committing their code.
4. Add a github action to reject PRs that introduce mixed line endings.

gshvang, 2021-09-30:
PR for the above fix has been merged. Closing the issue. Please reach out to us if you have any other questions.

closed 2021-09-30 by gshvang