FreeRTOS/FreeRTOS-Kernel #347: [Feature Request] Omit configTOTAL_HEAP_SIZE in Declaration of ucHeap[] when Using an Application Allocated Heap

Finwood, 2021-06-08:
**The problem**
In my microcontroller application (ARM Cortex-M4, 32K RAM) I would like to allocate all remaining RAM to FreeRTOS' heap. This could be achieved in the linker script:

```
SECTIONS {
    /* ... */

    .rtos_heap : {
        . = ALIGN(8);
        __FreeRTOSHeapBase = .;
        PROVIDE(ucHeap = __FreeRTOSHeapBase);
    } > RAM

    /* ... */

    __FreeRTOSHeapTop = __StackLimit;
}
```

I intended to use these addresses from inside `FreeRTOSConfig.h` like below:

```c
/* These are defined in the linker script */
extern const uint8_t __FreeRTOSHeapBase;
extern const uint8_t __FreeRTOSHeapTop;

/* Memory allocation related definitions. */
#define configAPPLICATION_ALLOCATED_HEAP 1
#define configTOTAL_HEAP_SIZE \
    ( (size_t) ( (uint32_t)&__FreeRTOSHeapTop - (uint32_t)&__FreeRTOSHeapBase ) )
```

That failed, because the `configTOTAL_HEAP_SIZE` macro referenced in the declaration below is not constant at compile time.

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/eec42331b4e7b83077f6c36fd0839d4057f904bd/portable/MemMang/heap_4.c#L64

**The solution I'd like**
The compiler (or at least _my_ GCC with _my_ set of compile flags, I'm not sure 🤔) does not need a compile-time constant value to declare a reference to an external array, for the compiler does not have to assign memory to it. Therefore, removing `configTOTAL_HEAP_SIZE` from the external declaration would work:

```c
/* Allocate the memory for the heap. */
#if ( configAPPLICATION_ALLOCATED_HEAP == 1 )

/* The application writer has already defined the array used for the RTOS
* heap - probably so it can be placed in a special segment or address. */
    extern uint8_t ucHeap[  ];
#else
    PRIVILEGED_DATA static uint8_t ucHeap[ configTOTAL_HEAP_SIZE ];
#endif /* configAPPLICATION_ALLOCATED_HEAP */
```

The other location `configTOTAL_HEAP_SIZE` is used in `heap_4.c` is inside `prvHeapInit(void)`, which does not need to be known at compile time.

https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/eec42331b4e7b83077f6c36fd0839d4057f904bd/portable/MemMang/heap_4.c#L349

mingyue86010, 2021-06-11:
Thank you for pointing out this using case. Discussion of changes can be see from the PR:

https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/351

mingyue86010, 2022-01-10:
As per discussion in PR: #351, this can be done by using heap_5. Check the detailed solution in the comment of #351. Please feel free to reopen the issue if you have any further question.

closed 2022-01-10 by mingyue86010