FreeRTOS/FreeRTOS-Kernel #358: [BUG] C90 compatibility broken since release 10.4.4

Vbrzeski, 2021-06-22:
**Describe the bug**
The Kernel fails to compile with C90 due to block scope on the _vEventGroupDelete()_ function of file _event_groups.c_.

**Target**
- Development board: Custom SBC
- Instruction Set Architecture: C166v2
- IDE and version: Keil uVision 5.29.0.0
- Toolchain and version: C166 v7.57

**Host**
- Host OS: Windows 10
- Version: 1909

**To Reproduce**
- Compile FreeRTOS with a C90 compiler

**Expected behavior**
Succesful compilation

**Screenshots**
[image]

**Additional context**
Looks like issues in the past may have introduced this bug: https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/293

**Analysis & Temporary Resolution**
The contract broken with the C90 standard is that variables must be declared at the top of a context block. If you're testing with GCC you may need to use the `-pedantic` flag to be strictly compliant. From my understanding thus far FreeRTOS still supports the C90 standard (of minus a custom stdint.h) so using the pedantic flag for your tests would be more correct.

I like to live at head, so I will just do a quick hack by placing the function in its own block context, thus the assertion can be done at the top. Here's a diff:
[image]

Again, thanks for the great kernel!

aggarg, 2021-06-22:
Would the following solve your problem:

```
    EventGroup_t * pxEventBits = xEventGroup;
    const List_t * pxTasksWaitingForBits;

    configASSERT( xEventGroup );

    pxTasksWaitingForBits = &( pxEventBits->xTasksWaitingForBits );
```

Vbrzeski, 2021-06-22:
Compiles! Thanks.

aggarg, 2021-06-23:
I merged the fix. Thank you for reporting the issue.

closed 2021-06-23 by aggarg