FreeRTOS/FreeRTOS-Kernel #365: [BUG]move the issue of https://github.com/FreeRTOS/FreeRTOS/issues/638 to this repo

WinnieeBear, 2021-07-08:
issue https://github.com/FreeRTOS/FreeRTOS/issues/638 on repo FreeeRTOS should be moved to here.

WinnieeBear, 2021-07-08:
I have fixed this issue, but I have no right to push the commit.

ravibhagavandas, 2021-07-09:
Hello,

Thanks for bringing up this and suggesting a fix for the issue. Could you create a pull request for the fix  by:
1. Creating your fork for this repository (steps can be found [here](https://docs.github.com/en/get-started/quickstart/fork-a-repo#forking-a-repository))
2. Creating a branch in that repository with all your changes.
3. Create a pull request from your branch to the main branch of this repository. (Steps can be found [here](https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request#creating-the-pull-request).

We will review the pull request and merge this once reviewed.

aggarg, 2022-07-25:
@WinnieeBear I am closing this issue. Feel free to raise a PR if you want to share your fix. 

Thanks.

closed 2022-07-25 by aggarg