FreeRTOS/FreeRTOS-Kernel #370: [Feature Request] RISC-V SMP Support

cmdrf, 2021-07-21:
**Is your feature request related to a problem? Please describe.**
In my Application I need to utilize both cores of the RISC-V MCU (Kendryte K210) because I need a lot of processing power. I attempted to do an SMP port for RISC-V myself (https://github.com/cmdrf/FreeRTOS-Kernel), but progress is very slow.

Since the [blog post](https://www.freertos.org/2021/06/introducing-the-freertos-symmetric-multiprocessing-smp-github-branch.html) mentions RISC-V, I wanted to ask if there is anything in the works already.

**Describe the solution you'd like**
Official RISC-V SMP support.

**Describe alternatives you've considered**
- There is a FreeRTOS fork for the K210 which supports multiprocessing, but it has been abandoned for a few years and is now outdated.
- A different RTOS with existing RISC-V SMP support?

**Additional context**
I have now successfully ported my SMP branch to QEMU, because JTAG debugging on the K210 is extremely unstable. Using QEMU makes development accessible to everyone.

mcergun, 2022-05-26:
seconding this

wangyuliu, 2023-12-05:
Hello everyone! I am working on implementing a tool to assess the complexity of CPU architecture porting. It primarily focuses on RISC-V architecture porting. In fact, the tool may have an average estimate of various architecture porting efforts.My focus is on the overall workload and difficulty of transplantation in the past and future,even if a project has already been ported.As part of my dataset, I have collected the **freeRTOS** project. **I would like to gather community opinions to support my assessment. I appreciate your help and response!** Based on scanning tools, the porting complexity is determined to be moderate, with a moderate amount of code related to the CPU architecture in the project.  Is this assessment accurate?Do you have any opinions on personnel allocation and consumption time？ I look forward to your help and response.