FreeRTOS/FreeRTOS-Kernel #389: [BUG]

dry-75, 2021-09-10:
**Describe the bug**
Stream buffer 's configMESSAGE_BUFFER_LENGTH_TYPE is not portable between little / big endian.

**Target**
- Development board: TMS570xxxx
- Instruction Set Architecture: ARMv7,  BE
- IDE and version: TI ARM CC v20.x.x
- Toolchain and version:  TI ARM CC v20.x.x

**Host**
- Host OS: Linux x86_x64
- Version: Fedora 32

**To Reproduce**
Configure FreeRTOS.h with #define configMESSAGE_BUFFER_LENGTH_TYPE uint8_t.  Send a message with xMessageBufferSend,
to another thread which is waiting for it on xMessageBufferReceive;  The receiver will now spin always getting 0 from the receive, as length of message is always 0, yet data is available in buffer..

**Expected behavior**
It should receive the exact sent message , once.

**Additional context**

In stream_buffer.c: in  the function : 
static size_t prvWriteMessageToBuffer( StreamBuffer_t * const pxStreamBuffer,
                                       const void * pvTxData,
                                       size_t xDataLengthBytes,
                                       size_t xSpace,
                                       size_t xRequiredSpace )

....
        ( void ) prvWriteBytesToBuffer( pxStreamBuffer, ( const uint8_t * ) &( xDataLengthBytes ), sbBYTES_TO_STORE_MESSAGE_LENGTH );
...

This is not portable between little vs big endian  With configMESSAGE_BUFFER_LENGTH_TYPE  uint8_t  it will write wrong byte on big endian. 

The messaging works if the type fixed to 32bits/4bytes, on this system

dry-75, 2021-09-10:
Needs something like this to fix:

`configMESSAGE_BUFFER_LENGTH_TYPE bufflen= xDataLengthBytes; `
`xShouldWrite = pdTRUE; `
`( void ) prvWriteBytesToBuffer( pxStreamBuffer, ( const uint8_t * ) &( bufflen ), sbBYTES_TO_STORE_MESSAGE_LENGTH ); `

archigup, 2021-09-10:
Hi, thanks for the bug report! Looking into it