FreeRTOS/FreeRTOS-Kernel #396: [Feature Request] allow the user reclaim memory after xTimerDelete

Kxuan, 2021-09-18:
**Is your feature request related to a problem? Please describe.**

`xTimerCreateStatic` create a timer base on the user providered memory buffer.
And `xTimerDelete` is use to delete the timer.

But when could the user relaim the buffer after calling `xTimerDelete`?

If the user immediately reuse the memory, the system will crash. Because `xTimerDelete` does not remove the timer from timer list. It just send a generic command to timer task.

**Describe the solution you'd like**
We can check xTimerIsTimerActive in xTimerDelete.

Make `xTimerDelete` return when xTimerIsTimerActive give `pdFALSE`.

**Describe alternatives you've considered**
One alternative way is sync the user task and timer task using semphore for each timer. But this solution requires additional memory resource to store the semphore, and we have to carefully handle in the timer task loop.

The user may call xTimerDelete in timer callback, we have to avoid deadlock and avoid access already deleted timer (timer delete in callback and reuse in prvProcessReceivedCommands). 

And maybe we can just add some documentation, tell the user do not immediately reuse the memory. And give an code example to user, such as tell the user wait xTimerIsTimerActive becomes pdFALSE for those timer created by xTimerCreateStatic.

**Additional context**
NONE

dachalco, 2021-09-22:
Hey @Kxuan 

Looking into this. Will try to get back to you later tonight or tomorrow morning.

dachalco, 2021-09-23:
@Kxuan 

Checking in. I've come up with some hacky ways to do this, but for such a use case I'd prefer something more formal. Discussing with Richard in the background. Will follow up again tomorrow.

Kxuan, 2021-09-23:
> ally, is there a particular reason you want the static mem reusable immediately after `xTimerDelete`? The changes you are suggesting would considerably impact timing characteristics of existing applic

@dachalco Thanks for your solution.

It is because a bug from a chip vendor's SDK. And that bug is related to this feature request.

The vendor's SDK provide its owner timer API. The API wraps FreeRTOS's `StaticTimer_t` into its structure `StaticTimerBuffer`. 
Even it named `Static***`, but the API providered to the user can only allocate timer dynamically.

When I don't need a timer, I call API to destroy it. Then, the SDK free the timer's memory after calling `xTimerDelete` immediately.

After that, our code may allocate memory using `malloc`, and fill the memory with some data. This write memory operation may crash the system. Because a `malloc` call may returns the memory region that was just release by timer.

Even worse, the timer task does not takes the highest priority. There are other tasks have higher priority (wifi/ble task) in the SDK.

Only do a loop on `xTimerIsTimerActive` may not solve this problem. May be the SDK code could add a `vTaskDelay(1)` in the `while` loop, or replace `StaticTimer_t` with a dynamically allocated `FreeRTOS` timer.

---

In any case, I think a method named `***Delete` shall release the resource it used when it returns success.
Or at least, the documentation should notice the user that `xTimerDelete` does not immediately release the memory.

So I open this issue, and hope this problem can be resolved.

RichardBarry, 2021-09-23:
Would it help if the memory used by the timer was memset() to 0 upon deletion?  That should be a simple change, although users would need to be aware of race conditions.  For example, it would be possible for two tasks to try and reuse the memory at the same time.

dachalco, 2021-09-23:
@Kxuan 

Oof. Thanks for the additional context, I see your dilemma. Interesting. So their provided `malloc` can return a static timer's memory (I'm hoping at least after it's been deleted)?

If you're able to modify `malloc` definition, perhaps you can add a check so it doesn't return the timer static memory if the timer is still active?

dachalco, 2021-09-23:
Added the requested info to the `xTimerDelete` documentation. It'll take some time to propagate to public site.

Kxuan, 2021-09-24:
> Would it help if the memory used by the timer was memset() to 0 upon deletion? That should be a simple change, although users would need to be aware of race conditions. For example, it would be possible for two tasks to try and reuse the memory at the same time.

Thank you @RichardBarry 

Currently, the function `xTimerIsTimerActive` could detected whether the asynchronous `delete` operation is done or not.
So the additional `memset` does not help much.

Kxuan, 2021-09-24:
> @Kxuan
> 
> Oof. Thanks for the additional context, I see your dilemma. Interesting. So their provided `malloc` can return a static timer's memory (I'm hoping at least after it's been deleted)?
> 
Yes.

> If you're able to modify `malloc` definition, perhaps you can add a check so it doesn't return the timer static memory if the timer is still active? This `malloc` should consider static timer memory unavailable if `xTimerIsTimerActive(...) == pdTRUE`. Not able to read said code, so let me know if I misunderstood something.

When the user requrest creating a timer with the SDK's API, it acturally dynamically allocates a `StaticTimerBuffer` using `malloc`, and returns a pointer to the allocated `StaticTimerBuffer`.
And because `StaticTimer_t` is a field on the structure `StaticTimerBuffer`, `StaticTimer_t` is dynamically allocated too.

Then, `xTimerCreateStatic` is called with the dynamically allocated `StaticTimer_t`.

So this is a dynamically allocated "static timer".

When the user request to destroy the timer, the SDK free the memory buffer using `free`. And then, the memory can be allocate again to other code.

---

The FreeRTOS documentation says:
```
If a software timer is created using xTimerCreateStatic() then the RAM is provided by the application writer, which requires an additional parameter, but allows the RAM to be statically allocated at compile time.
```
It does not disallow dynamically allocate the RAM. So it seems that it is a valid operation to create a static timer on dynamically allocated RAM.

dachalco, 2021-09-24:
@Kxuan 

> It does not disallow dynamically allocate the RAM. So it seems that it is a valid operation to create a static timer on dynamically allocated RAM.

I suppose you could, but I'm not sure why you'd want to or why it'd be necessary. Seems awkward. Was there feedback on this documentation reference? If so could you elaborate as I'm not sure I understand said feedback.

I really wish there was more I could do for you, but evidently these issues reside in the vendor SDK you've mentioned. Making changes to FreeRTOS to accommodate bugs in said SDK doesn't seem appropriate.

Considering the aforementioned documentation I added was mentioned as a suitable resolution to this issue, are you okay with closing this issue? You're always welcome to create new issues on new topics and I want to avoid diluting the scope of this thread.

Kxuan, 2021-09-24:
@dachalco Thanks. The documentation change is OK for this issue.

I have post this issue, because there is no API in FreeRTOS that releases the memory used by `StaticTimer` immediately. The users are unable to safely reclaim the memory after a timer never use again.

The name of `xTimerDelete` seems design to remove a timer. But acturally, it does not remove the timer. At least, when `xTimerDelete` returns success, the timer is still used by FreeRTOS timer task.

Some people (the chip vendor) do some wrong things, due to misunderstanding what `xTimerDelete` does.

It could be better that provide a formal way to release the memory used by `StaticTimer`.

But by adding some documentation for `xTimerDelete`,  those people can understand what `xTimerDelete` does. This solution makes developer `xTimerDelete` in correct way. And it is helpful too.

closed 2021-09-24 by Kxuan