FreeRTOS/FreeRTOS-Kernel #397: [BUG] vTaskDelayUntil(&xNextWakeTime, portMAX_DELAY) returns even though INCLUDE_vTaskSuspend is defined as 1

Stefan-Innerhofer, 2021-09-21:
**Describe the bug**
In my application I have a boot up task which handles the device boot up. After the boot up is completed, I put the task in block state with the following function.
`void BlockForEver(void)
{
	TickType_t xNextWakeTime = xTaskGetTickCount();
	// put this task in the blocked state for ever
	vTaskDelayUntil(&xNextWakeTime, portMAX_DELAY);

	//do some error handling
}`

portMAX_DELAY is defined as 0xFFFFFFFF, my system tick occurs every 200 µs and INCLUDE_vTaskSuspend is defined as 1. So the 0xFFFFFFFF system ticks are reached after after 9 days 22 h 36 min 33 s. Exactly after this time my error handlings hits.

As I understood so far vTaskDelayUntil(&xNextWakeTime, portMAX_DELAY); shall block a task indefinitely if NCLUDE_vTaskSuspend is defined as 1

**Target**
- Instruction Set Architecture: [ARM Cortex M4]
- compiler gcc 4.6.3
- FreeRTOS Kernel Version V10.4.2

**To Reproduce**
- configure system tick to 200 µs 
- define INCLUDE_vTaskSuspend as 1
- block a task with vTaskDelayUntil(&xNextWakeTime, portMAX_DELAY);
- error will occur after 9 days 22 h 36 min 33 s

**Expected behavior**
vTaskDelayUntil() shall not return

R-Zwi, 2021-09-21:
This code construct seem suspect to me - is there no coding rule against something like this:
[image]

RichardBarry, 2021-09-21:
Thanks for providing this feedback - however we always recommend to open a support request before a bug ticket.  In this case I can't see what is wrong with the code you circle, and I don't think that code will get executed in your case anyway.  The behaviour you describe is, I think, the expected behaviour as the task is unblocking at the time you asked it to - you cannot use vTaskDelay() or vTaskDelayUntil() to block forever as to do so would in effect create a zombie task.  I just checked [the documentation for vTaskDelayUntil()](https://www.freertos.org/vtaskdelayuntil.html) and it doesn't say portMAX_DELAY will block forever so I'm going to close this bug ticket.  If you still think there is a problem please ask a question on https://forums.freertos.org.

closed 2021-09-21 by RichardBarry

BerndThie, 2021-09-22:
Hello Richard, 
thanks for the fast reply!
I'm also interessted in this topic and found the following thread in th support forum: [https://www.freertos.org/FreeRTOS_Support_Forum_Archive/August_2012/freertos_Change_portMAX_DELAY_to_Lower_Value_5514046.html](url)

>Posted by richardbarry on August 10, 2012
>The behaviour of a block time of portMAX_DELAY depends on the setting of INCLUDE_vTaskSuspend() in FreeRTOSConfig.h.  If the configuration constant is set to 1, then portMAX_DELAY will cause the task to block indefinitely.  This means a function call that uses a block time of portMAX_DELAY need not check the return type of the blocking function call, because if the call returns, the task is guaranteed that the event it was waiting for has occurred. 

This post was from 2012, did the behavior of the RTOS change since then?

Thanks,
Bernd

RichardBarry, 2021-09-22:
I don't seem to be able to open that link for some reason - but as far as I know knowing changed.  If you are waiting for an event (queue message, semaphore, direct to task notification, whatever) then you can either wait for a fixed time or wait indefinitely.  If you delay for a fixed time, rather than wait for an event, then you unblock when that time has passed.  I'm not sure what it would mean to call vTaskDelay() or vTaskDelayUntil() with a parameter that says never wake up - you may as well delete the task.

BerndThie, 2021-09-23:
Hello Richard,
sorry for the dead link, my fault. I fixed it in the previous message.

There may be an workaround for the unexpected return of the infinite wait (eg. checking the return value an wait again), but is it working as designed?