FreeRTOS/FreeRTOS-Kernel #406: [BUG] ARM Cortex-A GIC cpu interface can not access by memory map

thomas-coding, 2021-11-03:
**Describe the bug**
 ARM Cortex-A GIC cpu interface can not access by memory map

From:
freertos/FreeRTOS/Source/portable/GCC/ARM_CA9/portASM.S
freertos/FreeRTOS/Source/portable/GCC/ARM_CA9/port.c
freertos/FreeRTOS/Source/portable/GCC/ARM_CA9/portmacro.h

It define GIC CPU interface as address like:
#define portICCIAR_INTERRUPT_ACKNOWLEDGE_REGISTER_ADDRESS 	( portINTERRUPT_CONTROLLER_CPU_INTERFACE_ADDRESS + portICCIAR_INTERRUPT_ACKNOWLEDGE_OFFSET )

But now, For GICv3 GIC-600,  Access IAR must like this:
asm volatile("mrc p15, 0, %0, c12, c12, 0" : "=r" (irq_number));

The way access GIC cpu interface by memory mapping is NOT supported now.
IHI0069G_gic_architecture_specification.pdf  
1.3.2 System register configuration
When affinity routing is enabled for execution in both Security states, the GIC must be configured to use System
register access to handle physical interrupts

Is there an update for this? Or any suggestions.
Thanks.

**Target**
- Development board: ARM Cortex-A 32
- Instruction Set Architecture: ARCH32
- IDE and version: 
- Toolchain and version: 

**Host**
- Host OS:
- Version:

**To Reproduce**

**Expected behavior**

**Screenshots**

**Additional context**

aggarg, 2021-11-21:
This is more of a feature request for new GIC which does not have memory mapped interface for CPU. Thank you for your post.

aggarg, 2022-11-14:
This is fixed here - https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/GCC/ARM_CA53_64_BIT_SRE

closed 2022-11-14 by aggarg