FreeRTOS/FreeRTOS-Kernel #419: [BUG] xQueue FromISR() APIs can cause an assertion based on RxLock and TxLock

preetpalkang, 2021-12-02:
**Describe the bug**

Assertion is hit in FreeRTOS 10.4.3: `configASSERT( cRxLock != queueINT8_MAX );`. This assertion was added after FreeRTOS 10.2.1

When `xQueueReceiveFromISR()` is used from inside of an interrupt, and we dequeue 128 items, the function can cause an assertion if `portYIELD()` is not invoked within the ISR. This behavior is not documented well, in fact, the documentation of `xQueueReceiveFromISR()` states the following:

```
From FreeRTOS V7.3.0 pxHigherPriorityTaskWoken is an **optional** parameter and can be set to NULL.
```

**This is terrible** because if we actually don't use `pxHigherPriorityTaskWoken`, then the FreeRTOS assertion can hit. It is therefore definitely NOT an "optional" parameter.

Initially, I investigated the logic of the Queues, and figured that the `rxLock` variable can cause race conditions. There seems to be somewhat an arbitrary choice of using `int8_t` for `rxLock` type. Due to this choice, if we get to a condition when this variable increments to `127`,  then the assertion is hit. From what I understand, this variable tracks how many tasks that might have gotten unblocked. In the ancient versions of FreeRTOS, it was used more like a flag to invoke `xTaskRemoveFromEventList()`. Then, it was used in a loop to invoke `xTaskRemoveFromEventList()`, and finally near FreeRTOS 10.4.3, there was an assertion added which is the core issue of this Github ticket.

See the code below on how the bug can be reproduced.

**Target**
- Development board: [e.g. HiFive11 RevB]
- Instruction Set Architecture: Cortex-M4
- IDE and version: VSCode
- Toolchain and version: 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

**Host**
- Host OS: MacOS (ARM CM4 program is cross-compiled)
- Version: Big Sur 11.6

**To Reproduce**

Run the following code that sets up a 1ms interrupt which dequeues 128 characters from a character Queue:

```c
static QueueHandle_t char_queue;

static void timer_isr(void) {
  char c = 0;
  const uint32_t threshold_for_freertos_assert = 128;

  BaseType_t higher_priority_task_woke = 0;
  for (int count = 0; count < threshold_for_freertos_assert; ++count) {
    xQueueReceiveFromISR(char_queue, &c, &higher_priority_task_woke);
  }

  // *************
  // **** BUG: If portYIELD() is not invoked, we hit the FreeRTOS assertion
  // *************

  if (higher_priority_task_woke) {
    // portYIELD();
  }

  hw_timer__ack_interrupt(); // This is not relevant to the problem; it just ACKs the timer ISR
}

static void test_task(void *p) {
  char c = 0;
  while (1) {
    xQueueSend(char_queue, &c, portMAX_DELAY); // Send data without any task sleep to reproduce problem faster
  }
}

/* ****************************************************************************
 */

int main(void) {
  char_queue = xQueueCreate(10000, sizeof(char));

  hw_timer__setup_interrupt(); // Assuming this function invokes 'timer_isr()' every 1ms

  const size_t stack_size_bytes = 1000 / sizeof(void *);
  xTaskCreate(test_task, "test_task", stack_size_bytes, NULL, PRIORITY_LOW, NULL);

  vTaskStartScheduler();
  return 0;
}
```

**Expected behavior**

- If `BaseType_t *pxHigherPriorityTaskWoken` parameter is not used along with `portYIELD()`, it should not cause an assertion
- Behavior of the usage of `BaseType_t *pxHigherPriorityTaskWoken` and invoking `portYIELD()` should be documented better
- Even if the user does not call `portYIELD()` inside of an ISR, it should not cause FreeRTOS to hit an assertion.

In general, APIs should guard themselves. If utilizing `pxHigherPriorityTaskWoken` is so important, then we should actually assert on this being NULL and it should not be an "optional" variable to the APIs. Furthermore, it is error-prone that even if the user used this variable, and did not invoke `portYIELD()`, it can cause the assertion.

Another solution is that the `rxLock` (or TxLock) should be guarded and be limited to the value of `queueINT8_MAX`. Even if these variables increment to their maximum values, it should gracefully unblock all the tasks waiting during `xQueueSend()` without being overly reliant on the value of `rxLock`.

**Screenshots**

If `configASSERT()` is defined, we will hit the following code:

```
configASSERT( cRxLock != queueINT8_MAX );
"cRxLock != queueINT8_MAX"
```

**Additional context**

N/A

jasonpcarroll, 2021-12-03:
Thank you for bringing this to our attention. The team is looking into it.

preetpalkang, 2021-12-03:
I modified the information. Previously, I could not reproduce the issue when I used `portYIELD()` from within the ISR, but that assumption is no longer true. I am able to reliably reproduce the problem.

RichardBarry, 2021-12-03:
> Assertion is hit in FreeRTOS 10.4.3: `configASSERT( cRxLock != queueINT8_MAX );`
Sorry this has been a problem for you.  It would be unusual to queue or dequeue that many items from within a single interrupt.  Doing so would be very inefficient, and there are primitives such as a stream buffer that are more suitable in most cases – although I’m not familiar with your particular application.  Hence I don’t really consider this a bug.  The assert is intended to help you by catching a potential issue should it happen, in this case an integer overflow.  As I recall the conversation at the time the assert was added concluded if the lock value became that high it was likely the interrupt had “run away” somehow too.  Are you able to use a stream buffer in place of the queue?  Or alternatively, use something like a task notification to unblock a task, then drain the queue from the task?

> From FreeRTOS V7.3.0 pxHigherPriorityTaskWoken is an **optional** parameter and can be set to NULL.

As per most things in FreeRTOS, the update was made in direct response to multiple end user requests – it is cleaner for those users not requiring the last parameter to not have to declare it and pass it in.  In that case any context switch that may be signaled as necessary by the last parameter would only happen when the currently running task blocked, or the tick interrupt executed, which ever came first.

> There seems to be somewhat an arbitrary choice of using `int8_t` for `rxLock` type
An 8-bit type is used because each there are two locks per queue (and therefore also semaphores), so the RAM consumed by the locks is 16-bits – whereas if a 32-bit type were used the RAM consumed by each queue and semaphore would be 64-bits.  We could update to use an unsigned value, giving a maximum lock count of 254, but at the expense of slightly more inefficient code and there would still be a limit at a relatively small number – albeit twice what it is now.

> Even if these variables increment to their maximum values, it should gracefully unblock all the tasks waiting during `xQueueSend()` without being overly reliant on the value of `rxLock`.

That too would be potentially much more inefficient if there we many tasks waiting.

I will take a new look at this to see if we can do something to make this better – but I would still not encourage users to queue or dequeue that many items from a single ISR.  For now though I'm not going to categorise this as a bug as there will always be a limit unless we rearchitect more fully.

preetpalkang, 2021-12-03:
@RichardBarry I have been following you for 10+ years, and I have also been teaching FreeRTOS for about 10+ years, and I will try to convince you that this is actually a **serious issue**.

This issue is hitting one of my clients who is blocked on factory release due to this problem. We have 100K+ vehicles basically waiting for an immediate solution. The issue in the firmware stemmed from this Queue assertion during final testing. The use case is a simple UART driver that actually does something very ordinary:
```
void uart_transmit_complete_isr(void) {
  const size_t hw_fifo_size = 16;
  for(size_t count = 0; count < hw_fifo_size; ++count) {
    if (xQueueReceiveFromISR(...)) {
      load_byte_to_uart_tx_fifo(byte);
    }
  }
}
```

When I convinced myself that the `cRxLock` is causing assertions, I dug down and tried to find its history. What is not clear to me is the significance of its count. Does this count represent the maximum amount of tasks that may have become unblocked? Getting a clear answer here would lead towards possible solutions and collaboration about this problem.

My next step after noticing that this assert is causing firmware failures was to reproduce the problem. The code you see above was an attempt to reproduce the problem. You may argue that an interrupt should not dequeue 128 bytes of data, but it can actually occur in nominal, sunny-day scenarios.

1.
First and foremost, if the `xQueueReceiveFromISR()` can fail after invoking it 128 times inside of an ISR, then this behavior needs to be clearly expressed and documented. Users should not get surprised about what they can and cannot do. Furthermore, the APIs should themselves guard themselves from misuse as much as possible. In general, misuse should not be possible, and if they can be misused, then they should have some sort of robust behavior.

2.
Secondly, this assertion is GUARANTEED to occur in nominal, sunny-day scenarios. Consider the following situation:
a. Task-A enters a section when `prvLockQueue()` occurs
b. Task-B, Task-C, Task-N get unblocked which are higher priority and keep running
c. Meanwhile, 200 bytes are loaded to be transmitted from UART
d. ISR keeps firing periodically to send 16-bytes at a time
e. Assertion will soon occur because the `cRxLock` will definitely increase to 127+

Now, to answer your points:
a) `It would be unusual to queue or dequeue that many items from within a single interrupt. Doing so would be very inefficient...` There is nothing unusual... modern CPUs have a lot of horsepower, and dequeuing 100+ bytes is perfectly fine... and it wouldn't even take 10uS to do so.

b) `Hence I don’t really consider this a bug.` Please don't be casual about this. This has hit production firmware to be landed on 100K+ vehicles per year. The entire reason why I'm here to report this is to collaborate with you and actually get a fix. Again, the naive code above is proof to the problem, but we have actually already hit the problem.

c) `As I recall the conversation at the time the assert was added concluded if the lock value became that high it was likely the interrupt had “run away” somehow too.` Please see bullet 2) above... this is not a theoretical or a run-away situation. It is a simple UART driver trying to do nominal things.

d) `Are you able to use a stream buffer in place of the queue?` No... Queue is the right primitive for a UART driver

e) `Task Notification`... This would be a hack because we are using the Queues as intended unless you believe that Queues should not be used with Interrupts.

f) `As per most things in FreeRTOS, the update was made in direct response to multiple end user requests`. Understood and concur! The problem report led me down the wrong path to question `pxHigherPriorityTaskWoken` but that logic is fine.

e) `That too would be potentially much more inefficient if there we many tasks waiting.`. I do not fully understand the purpose of `cRxLock` but would like to learn more from you.

@RichardBarry please re-open this bug. I can guarantee that users are going to start to hit this assertion if they upgrade to latest version of FreeRTOS. Let me know if you'd like to get on a call with one of your engineers :) I have your email!

preetpalkang, 2021-12-03:
Please do let me know if any of my assumptions are incorrect though. I'm happy to consider using a stream buffer, but I'm now very conscious and want to ensure that none of our Queue usage would hit this problem. I'm just a bit shocked that this kind of failure can occur somehow... but if I mis-used any APIs, I'd love to learn how this problem can be avoided by using the queues correctly.

preetpalkang, 2021-12-06:
Any update @RichardBarry @jasonpcarroll ?

RichardBarry, 2021-12-07:
When a queue is locked you can add or remove items to/from the queue, but you cannot update the lists of tasks that are blocked on the queue to wait to either send or receive.  The lock counts keep track of the number of items added or removed from a queue while it is locked.  When you unlock the queue you unblock tasks up to a maximum count of the locks - so potentially one task per lock count.  It is not possible to create tasks from ISRs, so you could potentially cap the lock counts to the number of tasks in the system - but again that would make the code more inneficient.  First you would have to query the number of tasks, then test against that value on each iteration of the loop.  Anyway - we are looking at different options but would still urge you to look at stream buffer instead - you can read all the characters in one go, and the read is more efficient.

preetpalkang, 2021-12-15:
@RichardBarry 

1. When we add items to the queue, that doesn't necessarily mean that we have that many tasks that are blocked on it right? Is there a way to check the count of the blocked task list and only increment the counter by that much?

2. Is my assumption correct that by the time we lock the queue, there cannot be any other task that can preempt the task that has locked the queue? I see this at `queue.c` but wanted to confirm. If this is true, then it does not appear that we would have an ISR runaway situation or similar that could take the lock count to an unexpected high value because the ISR exit should go back to the task that will very soon unlock and resume the scheduler.

```c
BaseType_t xQueueGenericSend(...) {
  // ...
        vTaskSuspendAll();
        prvLockQueue( pxQueue );
}
```

3. I believe this Github ticket may be to mainly **update API documentation and examples**. For instance, for the `xQueueSendFromISR()`, the API docs do not indicate and do not have any advise that we should ensure we never enqueue large amount of data. For instance, the https://www.freertos.org/a00119.html docs basically have the following example with a while loop but we should at the very least advise the users that they should not enqueue more than 100 bytes of data per ISR?

```c
void vBufferISR( void )
{
char cIn;
BaseType_t xHigherPriorityTaskWoken;

    do
    {
        /* Obtain a byte from the buffer. */
        cIn = portINPUT_BYTE( RX_REGISTER_ADDRESS );

        /* Post the byte. */
        xQueueSendFromISR( xRxQueue, &cIn, &xHigherPriorityTaskWoken );
    } while( portINPUT_BYTE( BUFFER_COUNT ) );
}
```

aggarg, 2022-01-06:
We added a cap to both `cTxLock` and `cRxLock` members of the queue data structures to the number of tasks in the system - https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/435

I am closing this issue. Feel free to re-open if this does not solve your problem.

Thanks.

closed 2022-01-06 by aggarg