FreeRTOS/FreeRTOS-Kernel #423: [BUG]  when i run stm32f103** connect with smartcard,why it always printf" Error:..\FreeRTOS\portable\RVDS\ARM_CM3\port.c,378"

Guoina, 2021-12-15:
**Describe the bug**
When i run stm32f103** connect with smartcard, why it always printf" Error:..\FreeRTOS\portable\RVDS\ARM_CM3\port.c,378"

**Target**
- Development board: [stm32f103ret6]
- Instruction Set Architecture: [Cortex-M3]
- IDE and version: [uvision v5.23.0.0 ]
- Toolchain and version: [MDK-ARM PLUS version 5.23]

windows

**To Reproduce**
- Use project ... and configure with ...
- Run on ... and could observe ...

**Expected behavior**
it never happens

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

paulbartell, 2021-12-15:
@Guoina, A copy of the source code for that assertion is located over here: https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/V9.0.0/FreeRTOS/Source/portable/RVDS/ARM_CM3/port.c#L370 

The associated note is copied below:
```
	/* This is not the interrupt safe version of the enter critical function so
	assert() if it is being called from an interrupt context.  Only API
	functions that end in "FromISR" can be used in an interrupt.  Only assert if
	the critical nesting count is 1 to protect against recursive calls if the
	assert function also uses a critical section. */
```

Generally, we suggest that runtime assertions in debug builds cause the application to stop altogether rather than continue running.

In short, FreeRTOS is working as-designed and your driver is likely calling the [taskENTER_CRITICAL](https://www.freertos.org/taskENTER_CRITICAL_taskEXIT_CRITICAL.html) macro from an ISR where it should be calling the [taskENTER_CRITICAL_FROM_ISR](https://www.freertos.org/taskENTER_CRITICAL_FROM_ISR_taskEXIT_CRITICAL_FROM_ISR.html) macro instead.

Feel free to reopen this ticket if you have any further questions.

closed 2021-12-15 by paulbartell

RichardBarry, 2021-12-16:
As per the instructions shown when opening a github issue (screenshot below), please use the support forum for support requests (https://forums.freertos.org), and only open bugs when an issue has been determined to be a bug.  Thanks.

[image]

Guoina, 2021-12-16:
Thans for your reply,and sorry for my "mistake bug ",I will check my code carefully.