FreeRTOS/FreeRTOS-Kernel #442: [Feature Request] Initialization ulMaxPRIGROUPValue outside xPortStartScheduler()

kaamil1984, 2022-01-17:
**Is your feature request related to a problem? Please describe.**

I have problem with SEGGER SystemView integration with ARM_CM7 FreeRTOS. In short - SEGGER SystemView initlialization stucks on assert, because xPortStartScheduler is not initialized when I call it. I can't simply call it after the sheduler starts because portable/GCC/ARM_CM7/r0p1/port.c xPortStartScheduler() function is blocking function.

**Describe the solution you'd like**

Temporarily:
1.  I created new function in portable/GCC/ARM_CM7/r0p1/port.c and 
2. I moved this part of xPortStartScheduler() to new xPortInitPriGroupValue() function:

```
/* configMAX_SYSCALL_INTERRUPT_PRIORITY must not be set to 0.
	     * See https://www.FreeRTOS.org/RTOS-Cortex-M3-M4.html */
	    configASSERT( configMAX_SYSCALL_INTERRUPT_PRIORITY );

	    #if ( configASSERT_DEFINED == 1 )
	        {
	            volatile uint32_t ulOriginalPriority;
	            volatile uint8_t * const pucFirstUserPriorityRegister = ( volatile uint8_t * const ) ( portNVIC_IP_REGISTERS_OFFSET_16 + portFIRST_USER_INTERRUPT_NUMBER );
	            volatile uint8_t ucMaxPriorityValue;

	            /* Determine the maximum priority from which ISR safe FreeRTOS API
	             * functions can be called.  ISR safe functions are those that end in
	             * "FromISR".  FreeRTOS maintains separate thread and ISR API functions to
	             * ensure interrupt entry is as fast and simple as possible.
	             *
	             * Save the interrupt priority value that is about to be clobbered. */
	            ulOriginalPriority = *pucFirstUserPriorityRegister;

	            /* Determine the number of priority bits available.  First write to all
	             * possible bits. */
	            *pucFirstUserPriorityRegister = portMAX_8_BIT_VALUE;

	            /* Read the value back to see how many bits stuck. */
	            ucMaxPriorityValue = *pucFirstUserPriorityRegister;

	            /* Use the same mask on the maximum system call priority. */
	            ucMaxSysCallPriority = configMAX_SYSCALL_INTERRUPT_PRIORITY & ucMaxPriorityValue;

	            /* Calculate the maximum acceptable priority group value for the number
	             * of bits read back. */
	            ulMaxPRIGROUPValue = portMAX_PRIGROUP_BITS;

	            while( ( ucMaxPriorityValue & portTOP_BIT_OF_BYTE ) == portTOP_BIT_OF_BYTE )
	            {
	                ulMaxPRIGROUPValue--;
	                ucMaxPriorityValue <<= ( uint8_t ) 0x01;
	            }

	            #ifdef __NVIC_PRIO_BITS
	                {
	                    /* Check the CMSIS configuration that defines the number of
	                     * priority bits matches the number of priority bits actually queried
	                     * from the hardware. */
	                    configASSERT( ( portMAX_PRIGROUP_BITS - ulMaxPRIGROUPValue ) == __NVIC_PRIO_BITS );
	                }
	            #endif

	            #ifdef configPRIO_BITS
	                {
	                    /* Check the FreeRTOS configuration that defines the number of
	                     * priority bits matches the number of priority bits actually queried
	                     * from the hardware. */
	                    configASSERT( ( portMAX_PRIGROUP_BITS - ulMaxPRIGROUPValue ) == configPRIO_BITS );
	                }
	            #endif

	            /* Shift the priority group value back to its position within the AIRCR
	             * register. */
	            ulMaxPRIGROUPValue <<= portPRIGROUP_SHIFT;
	            ulMaxPRIGROUPValue &= portPRIORITY_GROUP_MASK;

	            /* Restore the clobbered interrupt priority register to its original
	             * value. */
	            *pucFirstUserPriorityRegister = ulOriginalPriority;
	        }
	    #endif /* conifgASSERT_DEFINED */
```

3. I call new xPortInitPriGroupValue() function before SEGGER initialization and it works well.

**Describe alternatives you've considered**

Starting SEGGER from some task, after the scheduler starts.

RichardBarry, 2022-01-17:
I'm afraid I don't understand the problem you are describing, which may be because I'm not familiar with the Segger start up sequence, but as written I'm confused by your description, especially the part about vTaskStartScheduler() being blocking.  I assume the assert is triggered because something is being done out of sequence, in which case the assert is valid - you didn't actually say which assert got triggered though.

In any case it is highly unlikely we would separate out part of the start up sequence as to do so would impact many thousands of users and make usage more complex - so I'm going to close this feature request for now.  Happy to discuss further on the support forum (https://forums.freertos.org) if you would like to provide more information on the route cause of your issue.

closed 2022-01-17 by RichardBarry

kaamil1984, 2022-01-17:
Thank you for response and maintaining this project Richard.

I'm sorry for confusing you, english is not my first language, I was also tired and a bit lost in all this when I was writing this issue. I'm OK with your approach to my issue. I probably wrote this issue too early.

It looks like Segger startup function expects `ulMaxPRIGROUPValue` to be set when it is called, but this is weird...

I guess something changed between FreeRTOS 202012 and 202112 and FreeRTOS Segger become incompatibile in some cases.

Additionally I messed up everything a bit because I was calling xPortStartScheduler() instead of vTaskStartScheduler().

p.s. I think SEGGER company should give FreeRTOS developers few kits of their tools for free. FreeRTOS + SEGGER is very powerful combination.