FreeRTOS/FreeRTOS-Kernel #511: ARMv8M: SysTick_Handler vs xPortSysTickHandler?

a-wilhelm, 2022-06-22:
In most ports the SysTick handler is defined as
```
void xPortSysTickHandler( void )
```
with FreeRTOSConfig.h mapping it to the CMSIS name
```
/* Definitions that map the FreeRTOS port interrupt handlers to their CMSIS standard names. */
#define xPortSysTickHandler SysTick_Handler
```

However in the ARMv8M port the handler uses the CMSIS name:
https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/90d920466ea2e768e0095ef5a93ff0a0833d9068/portable/ARMv8M/non_secure/port.c#L754

Directly using the CMSIS name in port.c prevents using an alternate handler with the standard name without making some sort of modification.

Is this naming intentional or should it be changed to xPortSysTickHandler?

Skptak, 2022-06-23:
Hey,
Thanks for reaching out about this. I've notified a member of the kernel team about this issue and we'll look further into the cause of the naming.
Thanks for your time,
Soren Ptak

aggarg, 2022-06-27:
After I opened a PR to change the names of these handlers to match v7-M ports, one of my colleagues pointed that it is not a backward compatible change. 

Since we do not want to break backward compatibility, I want to know your opinion on some the following alternatives:
* Change the name of the SysTick handler in the vector table to something other than `SysTick_Handler`. 
* Change the name of the FreeRTOS SysTick Handler in FreeRTOSConfig.h:
```
#define SysTick_Handler Unused_SysTick_Handler
```

a-wilhelm, 2022-06-27:
I appreciate that there is not a solution that maintains backward compatibility.

I don't have a strong opinion on the alternatives. I would likely use the 
`#define SysTick_Handler Unused_SysTick_Handler`
approach to keep the changes localized to FreeRTOS and avoid modification of unrelated (vector table) code.

aggarg, 2022-06-28:
Thank you. I will close this issue then and will also close the PR I opened. Feel free to reopen or create a new one if you still face some issue.

closed 2022-06-28 by aggarg

jnz86, 2024-09-24:
I have a slightly different take on this problem.

I am using STM32 Cube MX, it wants the `HAL_IncTick()` called as well as the FreeRTOS code. 

So, I can't define a function name to be "unused". I do need to call it. So I think my only option is to modify the vector table to call my own, and call FreeRTOS's CMSIS-named SysTickHandler() from there... Or add my STM32 HAL code to the SysTickHandler at my CM33's port.c file.

Correct? 

Also... FWIW.... I would make the backwards compat breaking change. This was a mistake and should be corrected with a note.

aggarg, 2024-09-25:
> I am using STM32 Cube MX, it wants the HAL_IncTick() called as well as the FreeRTOS code.

When you use STM32 HAL with FreeRTOS, you must use a different timebase for HAL tick. STMCube generates a warning also if you do not do that. Please change the HAL timebase to a different timer - https://forums.freertos.org/t/what-i-learned-about-stm32-hal-stop-mode-and-crash-in-freertos/10247/5.

jnz86, 2024-09-25:
Thanks @aggarg ! All fixed up.

Moved the timebase to TIMx. Removed/Unchecked the Systick NVIC entries so now HAL is not trying to define a systick_handler in the `stm32yyxx_it.c` file. Everything is working as expected now.

BillFleming, 2025-08-15:
Yesterday I did a full search and replace (in the portable directory) of the 3 functions to the CMSIS names and everything is working as expected. There also won't be differences now between say IAR vs GCC tool chains which happened before on the CM0 port with assembly rework. I will be able to swap in ThreadX now in this software, without changing vector files, and in some other projects that use ThreadX, (STM, Renesas, etc) I will be able to copy this FreeRTOS into them to switch to it without needing to change the vector files. As an interesting bonus/challenge on this MCU there isn't a Systick timer, so shown here is an example of using one that goes in the other counting direction, with Systemview. There is no build error if I remove the Systick from the vector table. No further rename of the CM0 port files was needed to get accurate ticks or Systemview counts either.

FYI in the UL 60730 areas (appliances that need self-test code + certifications like washing machines, refrigerators, water heaters) there is a concept that customers now want to have bootloaders and create software into 2 or 3 pieces. They can segregate parts such as the self-test, bootloader, and vector table into the permanent area, then just update the application separately as needed. The self-test area can have it's own unique checksum (that doesn't change) that gets sent to UL and IEC certification boards. Then if they update the app software (say because some part needed to change on the PCB) that piece has a separate checksum.

Some of these customers also might have multiple brands of MCUs in the same appliance and they might want to use the same FreeRTOS copy for all of them. It could be possible for example to use the same port file but one MCU has an IAR project, and another is GCC. So having different names between ports would require modifying the port files. So being able to use the generic name everywhere will help with that also. With this restructuring towards the safety certification (where customers shouldn't change the files in any way) having the same function names across all port files should really be considered. If this breaks backwards compatibility just make that release part of version 12.0. But as shown you just need #defines to restore compatibility.

[image]

defermelowie, 2026-01-20:
The inconsistent naming is also conflicting with the docs on [Low Power RTOS For ARM Cortex-M MCUs](https://www.freertos.org/low-power-ARM-cortex-rtos). It clearly states that "The FreeRTOS tick interrupt handler is called xPortSysTickHandler()" with the exception of some "FreeRTOS packages distributed by third parties" where `xPortSysTickHandler()` is being called `SysTick_Handler()`.