FreeRTOS/FreeRTOS-Kernel #515: [BUG] SMP FreeRTOS scheduler start critical section race condition

Dazza0, 2022-06-27:
## Description

- Core 0 has already started the scheduler (via `vTaskStartScheduler()`)
- Core 1 has not called `xPortStartScheduler()` yet
- Core 0 triggers a yield on core 1 (e.g., by creating a new task with core 1 affinity)
- If core 1 attempts to enter a critical section (e.g., via calling `xTaskGetSchedulerState()`), core 1 will assert [in the following line within `prvCheckForRunStateChange()`](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/13f034eb7480f11c9fc43e7b51a1e55656985bf4/tasks.c#L671):

```
configASSERT( pxThisTCB->xTaskRunState != taskTASK_YIELDING );
```

## Cause

- Core 0 calling `vTaskStartScheduler()` should assign one of the idle tasks as core 1's current task
- Core 0 triggering a yield will cause core 1's current task's state to be `taskTASK_YIELDING`
- When core 1 enters a critical section, it will call `prvCheckForRunStateChange()` because the scheduler has already been started by core 0
- In `prvCheckForRunStateChange()`, core 1 calls `portENABLE_INTERRUPTS()` expecting a context switch to occur, but core 1 cannot switch contexts because it has not called `xPortStartScheduler()` yet.
- No context switch occurs, thus core 1's current task's state is still `taskTASK_YIELDING`.

## Possible Fixes

### Restrict usage of FreeRTOS API before the core has started the scheduler

One way to avoid this race condition is restrict a core from calling any FreeRTOS API before the core has called `xPortStartScheduler()`. However this may be inconvenient if the port has particular routines that are used both before and after the scheduler has started, and that those routines call `taskENTER_CRITICAL()`.

### Each core has their own `xSchedulerRunning`

Each core could individually track whether they have started their scheduler as such:

```c
PRIVILEGED_DATA static volatile BaseType_t xSchedulerRunnings[ configNUM_CORES ] = { pdFALSE };
#define xSchedulerRunning xSchedulerRunnings[ portGET_CORE_ID() ]
```

However, since `vTaskStartScheduler()` is only called from one core (e.g., core 0), all other cores will need a separate function to set their own `xSchedulerRunnings[x]` variable to `pdTRUE` when starting the scheduler, such as a `vTaskStartSchedulerOtherCores()` function that is called by all other cores before they call `xPortStartScheduler()`.

aggarg, 2022-11-02:
> One way to avoid this race condition is restrict a core from calling any FreeRTOS API before the core has called xPortStartScheduler(). However this may be inconvenient if the port has particular [...]

This seems like a reasonable solution to me. Can you describe what routine are you calling before calling `xPortStartScheduler` which calls `taskENTER_CRITICAL()`?

Dazza0, 2022-11-07:
@aggarg 

> Can you describe what routine are you calling before calling `xPortStartScheduler` which calls `taskENTER_CRITICAL()`?

In our case, it is mostly resource allocation APIs (e.g., memory, timers, interrupt allocation). Before each core starts scheduling, they must each allocate various resources on their own core. The same resource allocation APIs (i.e., libraries) are called to allocate resources both before and after the scheduling starts. These resource allocation APIs internally call FreeRTOS APIs (namely `taskENTER/EXIT_CRITICAL()`).

Note: These allocation API can't conditionally call FreeRTOS API depending on whether the scheduler has started or not, because calling `xTaskGetSchedulerState()` has the same issue as well. If core 0 has already called `vTaskStartScheduler()`, but core 1 hasn't started scheduling, then core 1 calling `xTaskGetSchedulerState()` will think that the scheduler as already started.

I think the crux of the issue is that if each core starts scheduling at different times in SMP, then the scheduler has 3 states (i.e., not started, partially started, fully started) as opposed to the binary states in single core (i.e., not started vs started). Given this third state, I see the following approaches to resolving/working around this race condition.

## 1. Eliminating the "Partially Started" via a SW Barrier

The "Partially Started" state could be eliminated if each port can guarantee that all the cores start scheduling at the same time. This would involve the use of some barrier flags/bits where...

- Core 0 cannot call `xTaskGetSchedulerState()` until all other cores have reached a stage (e.g., Stage A) where all other cores have finished their initialization (i.e., will no longer call FreeRTOS API)
- All other cores will spin on Stage A until core 0 allows them to progress (somewhere inside `xTaskGetSchedulerState() -> xPortStartScheduler()`).

With this approach, any FreeRTOS API call from any core before `xTaskGetSchedulerState()` will operate correctly, because no cores have started scheduling.

However, the downside to this approach is that the SW barrier will negatively affect startup time. Short startup time is important when deep sleeping/hibernating (where waking up counts as a full startup).

## 2. Remove FreeRTOS API calls from any core before scheduling starts

This may be difficult as it seems some FreeRTOS APIs were designed to be called before scheduling starts (namely any API that internally makes it behavior conditionally depend on `xSchedulerRunning`. Current APIs I could find are...

  - `xTaskCreate()` (or other task creation API), `vTaskDelete()`
  - `vTaskCoreAffinitySet()`
  - `vTaskPreemptionEnable()`
  - `vTaskSuspend()`
  - `vTaskSuspendAll()`, `xTaskResumeAll()`
  - `xTaskGetSchedulerState()`
  - `vTaskEnterCritical()`, `vTaskExitCritical()`

## 3. Making `xSchedulerRunning` per core

This approach would change  `xSchedulerRunning` to an array per core (e.g., `xSchedulerRunning[ configNUM_CORES ]`) and would represent whether a particular core has started scheduling yet. This is the current workaround adopted in ESP-IDF.

However, there would be open questions as to how `xSchedulerRunning` per core would affect the APIs such as `xTaskResumeAll()`.

chinglee-iot, 2022-11-17:
Using FreeRTOS APIs at not started state in multiple cores could still be a problem. Before the scheduler started, taskENTER/EXIT_CRITICAL and vTaskSuspendAll()/xTaskResumeAll() are empty functions. For example, if pvPortMalloc is called in multiple cores before scheduler started, there won't be any mutual exclusion protection for shared variables like xFreeBytesRemaining.

Is it possible that we allocate resource in the following methods:
* Allocate the resource on only one core before scheduler started, which is the core that starts FreeRTOS scheduler.
* Allocate the resource when the core is running a FreeRTOS task.

aggarg, 2022-12-19:
We concluded this one that FreeRTOS APIs should not be called from secondary cores until the scheduler has been started on those cores (i.e. xPortStartScheduler has been called on those cores).

closed 2022-12-19 by aggarg