FreeRTOS/FreeRTOS-Kernel #519: [BUG] Exception in use of queue

Vandoul, 2022-07-01:
**Describe the bug**
When call xQueueReceive(myQueue, portMAX_DELAY) in task1, then call vQueueDelete(myQueue) in task2, the task1 is always waiting for myQueue's new data but myQueue has been deleted!!!

**Expected behavior**
vQueueDelete function should resume all task which are suspended on the queue, then xQueueReceive should return an error.

Vandoul, 2022-07-01:
Return from xQueueReceive, only two conditions, new data is arrived or timeout, so when I call vQueueDelete, xQueueReceive hold runing and queue handle is invalid, the memory space of queue handle can be used in other task.

RichardBarry, 2022-07-01:
This describes the documented behaviour so is more of a feature request than a bug.  Please use the support forum if you would like to discuss.