FreeRTOS/FreeRTOS-Kernel #522: [BUG]

ani2732, 2022-07-14:
**Describe the bug**
A concise description of what the aaw is.

**Target**
- Development board: [e.g. HiFive11 RevB]
- Instruction Set Architecture: [e.g. RV32IMAC]
- IDE and version: [e.g. Freedom Studio 4.12.0.2019-08-2]
- Toolchain and version: [e.g. riscv64-unknown-elf-gcc-8.3.0-2019.08.0]

**Host**
- Host OS: [e.g. MacOS]
- Version: [e.g. Mojave 10.14.6]

**To Reproduce**
- Use project ... and configure with ...
- Run on ... and could observe ...

**Expected behavior**
A concise description of what you expected to happen.
Aa here.
e.g. code snippet to reproduce the issue.
e.g. stack trace, memory dump, debugger log, and many etc.

johnrhen, 2022-07-21:
Hello ani2732,

Are you experiencing a problem? It appears that this issue was created accidentally.

johnrhen, 2022-07-25:
Closing the issue due to inactivity. Please reopen if you're experiencing a problem.

closed 2022-07-25 by johnrhen