FreeRTOS/FreeRTOS-Kernel #526: [Feature Request] Add interrupt priority checking to CM33 ports

johnrhen, 2022-07-20:
**Is your feature request related to a problem? Please describe.**
Interrupt priority checking is missing on CM33 ports.

**Describe the solution you'd like**
Update the CM33 ports to check interrupt priorities, in line with the other Cortex-M ports. The other Cortex-M ports check interrupt priorities using `configMAX_SYSCALL_INTERRUPT_PRIORITY` and use the function `vPortValidateInterruptPriority()` inside of the`portASSERT_IF_INTERRUPT_PRIORITY_INVALID()` macro.

**Additional context**
https://forums.freertos.org/t/cm33-missing-vportvalidateinterruptpriority/14836

If you have the same (or similar) feature request, please upvote this issue with thumbs up 👍
and use the comments section to provide answers to the questions above.

nilesh-dryad, 2023-04-18:
Is there any update on this enhancement?

urutva, 2023-05-15:
The PR https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/673 merged to main. Closing the issue.

closed 2023-05-15 by urutva