FreeRTOS/FreeRTOS-Kernel #531: [BUG] compatibility issue regarding xQUEUE struct

duchao713, 2022-07-28:
The _xQUEUE_ struct changed after [https://github.com/FreeRTOS/FreeRTOS-Kernel/commit/4fbcdbf13bf57a2363277b165ea6345fe8cb9e57](url). Which may cause some compatibility issue if the member of _xQUEUE_ is quoted, such as _pcTail_.
Is it possible to solve this compatibility issue using _configENABLE_BACKWARD_COMPATIBILITY_ ?

RichardBarry, 2022-07-28:
Those changes were made four years ago.  I recommend opening a support request thread on the support forum (https://forums.freertos.org) that describes the issue you are having as it's not clear what it means to have a structure member "quoted".  Hopefully we can find a resolution for you there.

duchao713, 2022-07-29:
> Those changes were made four years ago. I recommend opening a support request thread on the support forum (https://forums.freertos.org) that describes the issue you are having as it's not clear what it means to have a structure member "quoted". Hopefully we can find a resolution for you there.

Thanks for the reply.
I mean the structure member _pcTail_.
In some old FreeRTOS projects, pxQueue->pcTail is OK, but error will be reported if the FreeRTOS updated to a newer version.
This is the compatibility issue I mentioned.

RichardBarry, 2022-07-29:
Sorry - it is still not clear to me.  pcTail was a member of a private structure.  By design, application code can't use it, so we should be free to change it without breaking anybody's code.

duchao713, 2022-08-01:
I got the point and will close this.

closed 2022-08-01 by duchao713