FreeRTOS/FreeRTOS-Kernel #553: [BUG] Extra semicolon in vTaskNotifyGiveFromISR and vTaskGenericNotifyGiveFromISR

newbrain, 2022-09-05:
**Describe the bug**
Spurious semicolons at the end of vTaskNotifyGiveFromISR and vTaskGenericNotifyGiveFromISR macro definitions.
Version: FreeRTOS Kernel V10.4.3 LTS Patch 2 (still present on latest commit)

**Target**
- Development board: Immaterial
- Instruction Set Architecture: Immaterial
- IDE and version: Immaterial
- Toolchain and version: llvm, clang 13.0, (immaterial)

**Host**
- Host OS: Any
- Version:

**To Reproduce**
Compile the following code as part of an ISR:
```      BaseType_t yield; 
        if (condition)
            vTaskNotifyGiveFromISR(task, &yield);
        else
            xSemaphoreGiveFromISR(semaphore, &yield);
```
**Expected behavior**
Compilation should succeed.
it instead fails with:
```
[build] ../source/SDR/filterGeneric.c:97:9: error: expected expression
[build]         else
[build]         ^
[build] 1 error generated.
```

**Additional Context**
The problem is caused by extra semicolons at the end of vTaskNotifyGiveFromISR and vTaskGenericNotifyGiveFromISR [macro definitions:](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/6324f6fc3e26d8b550fe54d474cd1ea89e238622/include/task.h#L2533-L2536)
```
#define vTaskNotifyGiveFromISR( xTaskToNotify, pxHigherPriorityTaskWoken ) \
    vTaskGenericNotifyGiveFromISR( ( xTaskToNotify ), ( tskDEFAULT_INDEX_TO_NOTIFY ), ( pxHigherPriorityTaskWoken ) );
#define vTaskNotifyGiveIndexedFromISR( xTaskToNotify, uxIndexToNotify, pxHigherPriorityTaskWoken ) \
    vTaskGenericNotifyGiveFromISR( ( xTaskToNotify ), ( uxIndexToNotify ), ( pxHigherPriorityTaskWoken ) );
```
The semicolons terminate the statement and make them not follow the usual convention for function-like macros of behaving like an actual function, specifically making their use in a single statement `then` clause without braces unexpectedly _not_ require a semicolon.

jasonpcarroll, 2022-09-06:
Hi @newbrain,
I'll let the team know. I'd also suggest that you make a PR that fixes the issue and the team can review.
Best,
Jason Carroll

newbrain, 2022-09-07:
Hi, I'll try to do that this evening (Europe time)!
I'll also scan the other includes for similar issues: this one actually hit me, #554 I just found checking for similar stuff.

closed 2022-09-12 by paulbartell