FreeRTOS/FreeRTOS-Kernel #554: [BUG] VtaskDelayUntil() macro badly defined

newbrain, 2022-09-05:
**Describe the bug**
The definition of function-like macro vTaskDelayUntil does not follow usual conventions.

**Target**
- Development board: Irrelevant
- Instruction Set Architecture: Immaterial
- IDE and version: Any
- Toolchain and version: Any C standard compliant compiler

**Host**
- Host OS: Any
- Version: -

**To Reproduce**
Compile any code of the following form:
```
if (condition)
    vTaskDelayUntil(&prevTime, delay);
else
   /* Any statement or block */
```
It will fail with a message similar to:
```
[build] ../source/main.c:521:5: error: expected expression
[build]     else
[build]     ^
[build] 1 error generated.
```

**Expected behavior**
Compilation should succeed.

**Additional context**
The problem lies in the definition of `vTaskDelayUntil()` [function-like macro](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/6324f6fc3e26d8b550fe54d474cd1ea89e238622/include/task.h#L868-L871):
```
#define vTaskDelayUntil( pxPreviousWakeTime, xTimeIncrement )                   \
    {                                                                           \
        ( void ) xTaskDelayUntil( ( pxPreviousWakeTime ), ( xTimeIncrement ) ); \
    }
```
The block surrounding the call  to `xTaskDelayUntil` makes the macro a complete and terminated statement, needing omission of the terminating semicolon when used in a non-block, single statement `then` clause of an if statement.
The usual way to avoid this is surrounding the block in the macro with a `do {...} while(0)` to form a non terminated statement.
See also #553 .

jasonpcarroll, 2022-09-07:
Hi @newbrain,
Thank you for making us aware of this. If you could make a PR that the team could review that would be ideal.
Best,
Jason Carroll

paulbartell, 2022-09-08:
Closing this issue since #555 has been merged.

closed 2022-09-08 by paulbartell