FreeRTOS/FreeRTOS-Kernel #593: [Feature Request] Timers and pended functions should be separated

tcpluess, 2022-11-30:
**Is your feature request related to a problem? Please describe.**
Currently, the software timers also include the functionality of "pended functions". This is useful if one wants to defer the execution of lengthy or less important functions to the daemon task, i.e. the timer service task. While the functionality of these "pended functions" is very useful, having it combined with the timer service task has a couple drawbacks:

- if one wants to use the pended functions, the timer module must be included as well. Decoupling of these two functionalitites, which are in fact two completely different things, would be great.
- if software timers and pended functions are used at the same time, this can lead to problems if one of the pended function takes longer than the time required until the next timer is ready. Assume the following scenario: a software timer is configured that should fire in 2 ms. Further, a pended function is installed, whose execution would take 3 ms. In that case, the timer would be delayed. It would make sense to have a task whose priority is just above the idle task, and then defer lengthy operations to this task using the pended functions. The timer task then would have a priority that is higher than this, such that it will not be interrupted by long operations.

**Describe the solution you'd like**
There should be two separate tasks for the pended functions and the timers such that one can use either one alone, or both.
I have implemented this already and will post a PR if this is considered useful.

**Describe alternatives you've considered**
Use pended functions from the timer module.

**How many devices will this feature impact?**
N/A

**What are your project timelines?**
N/A

**Additional context**
none

If you have the same (or similar) feature request, please upvote this issue with thumbs up 👍
and use the comments section to provide answers to the questions above.

paulbartell, 2022-11-30:
@tcpluess Thanks for the feature request. I will add it to our list internally for prioritization. 

Feel free to open a pr and we can discuss your implementation further.

alfred2g, 2023-01-18:
As Suggested on the PR, please create it as a Demo not as a Kernel PR.
Thanks.